suaveolent/ha-hoymiles-wifi #1: Total Engery is missung

mimimi0815, 2023-12-29:
Hi , thanks for this cool Addon.
I have seen that total Energy is missing. For the single Ports it is already available. 
But for the total inverter it is missing (Total Engery)

culixnet, 2023-12-30:
can agree, great to have this integration now for inverters with built in wifi, thx a lot!
But TotalEnergy is missing. But from my point of view it would be easier to read to have the TotalEnergy in the unit_of_measurement in ENERGY_KILO_WATT_HOUR instead of ENERGY_WATT_HOUR. Same for TotalEnergy of the single modules.

suaveolent, 2024-01-03:
I am reading the values directly from the inverter. 

Unfortunately, the inverter does not provide the total energy, just daily energy. 

Of course I could calculate the total energy from the individual ports, however, this would not be accurate: 

The total energy of the ports is DC, the daily energy is AC. (I will probably rename this in the next versions).  So there will be a slight conversion loss (DC-> AC and the inverter consuming some of the power) and the values will not match.

If you want to integrate it into the energy dashboard, use the Daily Energy value. It works great since it resets every day.

For Kilo Watt Hours I think you could change that manually in Home Assistant.

culixnet, 2024-01-04:
tried to add total energy by myself, only to realize what you mentioned, it is not available... 
Also thx for the clarification of DC and AC between ports and inverter

switched the total_energy of ports successfull to kWh, thx for the hint, I am new to HASS

I think you could close this issue

closed 2024-01-05 by suaveolent