suaveolent/ha-hoymiles-wifi #2: Preserve last value when inverter becomes unavailable.

ordo01, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
Dashboards like Energy Flow Card Plus show wrong values when the inverter becomes unavailable and all sensors show unknown as value, e.g. after sunset. 

**Describe the solution you'd like**
Preserving the last value of sensors which act as counters like total energy would help. Just like it is done in the hoymiles-mqtt-publisher from DennisOSRM. Other sensors showing a current value like power should go to zero instead.
Plus having an additional sensor indicating the state available/unavailable would be appreciated.

suaveolent, 2024-01-09:
Thank you this is a good idea.

I will implement it in one of the upcoming versions.

closed 2024-01-09 by suaveolent

suaveolent, 2024-01-10:
> Thanks for the quick implementation.
> 
> I see a new release in GIT, but this does not appear on HACS. Will it just take time to get visible in HACS?

If HACS does not (yet) show the update you can navigate to HACS --> Integrations --> Hoymiles HMS-XXXXW-T2

Then on the top right click on the three dots and click on "Update information" , then "Redownload" and select the v0.0.5.

commit 9becca49 (2024-01-09): add restore sensor for state class energy. fixes #2