suaveolent/ha-hoymiles-wifi #3: AC Daily Energy was not reseted

mbrockeu, 2024-01-23:
**Describe the bug**
The AC Energy for daily usage is not reset to 0 at the beginning of the day.

**To Reproduce**
Steps to reproduce the behavior:
1. Open sensor at the beginning of the day when the Inverter starts producing

**Expected behavior**
The value will be set to 0 if this is also the value in the app.

**Screenshots**
Screenshot was taken at 8:57 compared to app the values are not correct
[image]

[image]

Version is 0.0.7

mbrockeu, 2024-01-23:
Ok forget what I say, after 1 Wh is produced the correct value is shown ;-)

suaveolent, 2024-01-23:
Great, closing this issue then.

closed 2024-01-23 by suaveolent

culixnet, 2024-01-29:
Had a similar effect: One day I accidentally reduced the AC output to 0% for a whole day -  HA then displayed and counted the Wh from the day before