suaveolent/ha-hoymiles-wifi #6: Reset daily energy to 0 at the start of the day.

ordo01, 2024-01-29:
It looks odd when the daily power delivery value on a dashboard shows high numbers already early in the morning. 
For an improved tracking it would be nice if the sensors for daily energy would be reset to zero at midnight and not when the DTU is delivering values again the following day.

suaveolent, 2024-01-30:
Thanks. This has been bugging me too.

I will try to implement a workaround by resetting the values at midnight.

suaveolent, 2024-01-30:
@ordo01  I have pushed a preliminary test to the main branch. You can test the feature if you want to by updating to main instead of release trough HACS.

I have not personally tested the feature yet, so it might not work as expected.

ordo01, 2024-01-31:
Thanks @suaveolent, I will give it a try later this week.

suaveolent, 2024-02-03:
@ordo01 Should be fixed with v0.1.0, please reopen if issue still persists

closed 2024-02-03 by suaveolent