suaveolent/ha-hoymiles-wifi #7: weired text in host configuration

clinke77, 2024-02-03:
First: thanks for all the effort.

On the configuration page I would expect to be asked for the Host IP address, instead I get asked for a weird string:
[image]

When I close the configuration I do not see my configured IP, instead I see:
[image]

Besides this I did not manage to make it work. I do manage to connect via the python commandline from my host PC to this IP address:

hoymiles-wifi get-real-data-new --host 192.168.178.136
3.6.1
Get-real-data-new Response: device_serial_number: "414392319514"
timestamp: 44994
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 22069994689812
  firmware_version: 1
  voltage: 2312
  frequency: 5001
  active_power: 251
  current: 10
  power_factor: 999
  temperature: 109
  warning_number: 100
  crc_checksum: 28864
  link_status: 1
  modulation_index_signal: 16646270
}
pv_data {
  serial_number: 22069994689812
  port_number: 1
  voltage: 4
  current: 1
  error_code: 50331648
}
pv_data {
  serial_number: 22069994689812
  port_number: 2
  voltage: 422
  current: 62
  power: 265
  energy_total: 297
  energy_daily: 34
  error_code: 50331648
}
dtu_power: 251
dtu_daily_energy: 34

anything I need to install? what should the logs look like? I only get a log when I update the config.

clinke77, 2024-02-03:
Just figured out how to get some logs:

2024-02-03 13:31:44.652 ERROR (MainThread) [custom_components.hoymiles_wifi] Options update called!
2024-02-03 13:32:19.169 DEBUG (MainThread) [custom_components.hoymiles_wifi] Hoymiles data coordinator update
2024-02-03 13:32:19.171 DEBUG (MainThread) [custom_components.hoymiles_wifi] Unable to retrieve real data new. Inverter might be offline.
2024-02-03 13:32:19.172 DEBUG (MainThread) [custom_components.hoymiles_wifi] Finished fetching hoymiles_wifi data in 0.003 seconds (success: True)
2024-02-03 13:32:19.172 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for {self.name) to avoid resetting total_increasing counter
2024-02-03 13:32:19.173 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for {self.name) to avoid resetting total_increasing counter
2024-02-03 13:32:19.173 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for {self.name) to avoid resetting total_increasing counter
2024-02-03 13:32:19.174 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for {self.name) to avoid resetting total_increasing counter
2024-02-03 13:32:19.174 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for {self.name) to avoid resetting total_increasing counter

suaveolent, 2024-02-03:
Thanks, the weird string results from preparations to merge into home assistant core. I will fix the string in the next update. 

I noticed also that you called the options update. There might be a separate issue with that, since home assistant needs to be restarted for them to be applied. 
Can you try setting up the integration again to see if it works?

suaveolent, 2024-02-03:
@clinke77 The weird text should be fixed now with v0.1.1

clinke77, 2024-02-03:
@suaveolent I can confirm that now there is a nice "Host" written in the new version. And after a reconfigure and a restart I got the same logs as above.
I also tried to  run the commandline tool from inside the homeassistant container and could successfully connect to the inverter. I also got the mqtt solution (https://github.com/DennisOSRM/hms-mqtt-publisher) to work but I think your approach is much more appealing.
Now sun is down and unlike any other tech stuff you cannot test this in night shifts so I have to postpone to tomorrow.

suaveolent, 2024-02-04:
@clinke77 it looks like the integration is using the ip and port from a completely different integration. Are you running the latest version of home assistant (2024.1.6)?

If not, can you try deleting the integration entry, updating to latest home assistant and then setting up the integration again?

closed 2024-03-31 by github-actions[bot]