suaveolent/ha-hoymiles-wifi #8: Cannot add custom repo

marcelK0, 2024-02-05:
**Describe the bug**
When I'm adding the custom repo according to the docs I get

_No RepositoryFile.MAINIFEST_JSON file found 'custom_components/hoymiles_wifi/RepositoryFile.MAINIFEST_JSON'_

**To Reproduce**
Steps to reproduce the behavior:
As described in the readme of the repo

**Expected behavior**
Custom repo to be added from github

**Screenshots**

**Desktop (please complete the following information):**
 - OS: OSX
 - Browser Safari
 - Version [e.g. 22]

suaveolent, 2024-02-06:
I tried with latest and home assistant core and could not reproduce the issue. Can you maybe provide some more log information from HACS install process?

marcelK0, 2024-02-06:
ok, looks like HACS needed an update. Might be worth mentioning, as HACS is never updated unless you manually run the installer from the terminal again. It seems versions from 2022 are broken with today's HA.
Also some instructions on your doc are outdated, eg. step 5, it never asks to 'save' or 'install', but 'download', etc.
cheers

marcelK0, 2024-02-06:
solved

closed 2024-02-06 by marcelK0

suaveolent, 2024-02-06:
Thanks for letting me know. Will update the Readme.