suaveolent/ha-hoymiles-wifi #9: Missing `crcmod` dependency and Config flow could not be loaded: {"message":"Invalid handler specified"}

feddischson, 2024-02-18:
**Describe the bug**

It seems that the `crcmod` python library dependency is missing.

**To Reproduce**
Steps to reproduce the behavior:
 1. follow installation procedure shown in the README.md

**Expected behavior**
 - No failure

**Actual behavior**
 - The message  `Config flow could not be loaded: {"message":"Invalid handler specified"}` is shown when adding the integration. 

**Setup**
 - homeassistant  2024.2.2
 - HACS 1.34.0
 - ha-hoymiles-wifi v0.1.2

**Additional context**
I was able to fix the issue by installtion `crcmod` manually (`pip3 install crcmod`)

suaveolent, 2024-02-18:
Thanks, I was assuming that the package would be installed automatically. I will investigate, maybe it must be fixed in the hoymiles-wifi lib.

suaveolent, 2024-02-19:
Should be fixed with `v0.1.3`

davkri19, 2024-03-29:
I have the Same Problem with v0.2.0 but the fix with crcmod was Not successfull. With v0.1.3 it Works.

davkri19, 2024-03-29:
The version is 2024.3.3 on a newly installed system.

suaveolent, 2024-03-29:
I see. Did manually solving crcmod solve the issue?

davkri19, 2024-03-29:
> I see. Did manually installing crcmod solve the issue?

No, unfortunatily not.

suaveolent, 2024-03-29:
Can you post your log to see what is going on?

davkri19, 2024-03-29:
Logger: homeassistant.config_entries
Source: config_entries.py:2444

Error occurred loading flow for integration hoymiles_wifi: No module named 'hoymiles_wifi.inverter'

suaveolent, 2024-03-29:
That seems weird. The `hoymiles_wifi.inverter` import is from an older version. The new import should be `hoymiles_wifi.dtu`

You can check the `https://github.com/suaveolent/ha-hoymiles-wifi/blob/main/custom_components/hoymiles_wifi/config_flow.py` file and compare it to your local install.

How did you upgrade to `v0.2.0`?

davkri19, 2024-03-29:
All local Files are exactly the Same as the files in GitHub. I cant find hoymiles_wifi.inverter in any file.
For Installation I tried Both ways in Hacs and manual

suaveolent, 2024-03-29:
Have you tried completely restarting Home Assistant manually? I remember when testing accounting a similar issue when restarting through HACS.

suaveolent, 2024-03-29:
Also you can try manually upgrading the hoymiles_wifi library via pip3 install —upgrade hoymiles_wifi command

davkri19, 2024-03-30:
I found my stupid mistake. I downladed an older Version into the root folder and the integration used that folder instead of the folder under custom_components.
Now it works Perfectly
Thank you for your help and this Great work.

suaveolent, 2024-03-30:
Great! Glad it works!