suaveolent/ha-hoymiles-wifi #11: Upgrade to version 0.1.3 delivers enormous DC 1 power values

culixnet, 2024-02-20:
In my dashboard I have a statistic graph which visualize the Power of every Port in one diagramm which worked very finde for me. But since the upgrade to version 0.1.3 I get some strange peaks with the value of 27.061.504,1 W. Screenshot:

[image]

I am sure that these values are not correct ;-) Any ideas?

Thx a lot
Mario

suaveolent, 2024-02-20:
That is indeed very weird. 
I could not observe the same behavior with my DTU. 

Which DTU version are you on? (Should be displayed with newest version) 

is there something in your logs which might explain the behavior?

domhaas, 2024-02-20:
Got also this issue, but only on AC. DTU is running on V00.01.11:

[image]

[image]

Inverter is a 800 and currently running with only 1 Panel (440W) - tomorrow I got my second Panel.
I will give you feedback if this happens again.

p1ngod, 2024-02-20:
Got the same since the last update, but only once on Port 1. DC power was exactly the same (27.061.504,1 W), DC current 100,08A.

Maybe a problem with the async update in hoymiles-wifi?

HMS-800W-2T
DTU Firmware V00.01.11, Hardware H00.01.00
Inverter Firmware V01.00.08, Hardware H00.04.00

suaveolent, 2024-02-20:
Thanks for letting me know. 
Maybe there is a transmission error? I will implement the crc check also for the response and see if that solves the problem or if there is some kind of issue with the async calls.

culixnet, 2024-02-20:
> Which DTU version are you on? (Should be displayed with newest version)
>
HW Version of DTU: 00.01.00
SW Version of DTU: 00.01.08

There exists a 00.01.11 SW version, but I did not install it because of a missing changelog.

> Is this also true for port2?
> 
nope, only port_1_power and port_1_current (100,08A)

> is there something in your logs which might explain the behavior?

only that: `Entity sensor.hoymiles_hms_800_port_1_daily_energy from integration hoymiles_wifi has state class total_increasing, but its state is not strictly increasing. Triggered by state 760 (768.0) with last_updated set to 2024-02-20T15:02:06.388178+00:00. Please report it to the author of the 'hoymiles_wifi' custom integration`
That`s another entity, but perhaps as consequence of the other error?
Is there an additional logfile where I should have a look?

domhaas, 2024-02-21:
@suaveolent It seems like the issue occures when the last fetch was not successful:

[image]

Couldn't find anything in the logs. Maybe the crc-check for the response could help here.

suaveolent, 2024-02-21:
Thanks for letting me know. I noticed that sometimes the response is invalid. Maybe because the inverter cannot handle all the different requests at the same time.  (Even though I already use different timings, but eventually there will be some overlap).
I will evaluate if there is some way, e.g. mutex to ensure that only one request is issued at a time.

suaveolent, 2024-02-21:
Might already be fixed in main with fixes for #12 

Can you maybe update to main to check if this fixes your problem?

culixnet, 2024-02-21:
> Might already be fixed in main with fixes for #12
> 
> Can you maybe update to main to check if this fixes your problem?

I deployed from main at 13:00 CET, since then I had no peak again. Will observe it tomorrow and the next days

Jonas242, 2024-02-22:
Same Issue - happens only for one DC Channel (Channel 1), the reported value is always the same: 27061504,1 W
and maybe another issue of the same type:
- AC Power sometimes has a value of 0 within the day

[image]

culixnet, 2024-02-22:
> > Might already be fixed in main with fixes for #12
> > Can you maybe update to main to check if this fixes your problem?
> 
> I deployed from main at 13:00 CET, since then I had no peak again. Will observe it tomorrow and the next days

Also today no peak with the main branch

suaveolent, 2024-02-24:
`v0.1.4` just released, please reopen issue if problem still persists

closed 2024-02-24 by suaveolent