suaveolent/ha-hoymiles-wifi #12: S-Miles Cloud stopped pulling data

esommerfeld, 2024-02-21:
the integration in HA works great, but since then the s-miles cloud no longer retrieves any data. is it somehow possible to use both? thank you!

suaveolent, 2024-02-21:
Thanks for the report. You are right. Might be due to too many requets. I will try to increase the time of certain functionality (e.g. sw. version) and see if it fixes the problem.

esommerfeld, 2024-02-21:
Weird: In reality, S-Miles retrieves the data every 15 minutes, but doesn't update the time. :D

suaveolent, 2024-02-21:
Already found the bug. I need to increase the interval to retrieve the Diagnostics data of the DTU (e.g. HW / SW ) version. Then update of app is working again. Will fix with next release (or update to main)

closed 2024-02-21 by suaveolent