suaveolent/ha-hoymiles-wifi #13: Cannot Download in Custom Repositories

Kikash10, 2024-02-22:
Hey!

I apologise  for creating new bug report because it perhaps does not belong here but i didn't know where else to request little help..
Following your installation guidance, I come to the point where I have added your link to Custom Repositories. But clicking on it or doing anything else never provides a DOWNLOAD option. I tried rebooting and all of that but I simply cannot find your custom repo anywhere.
Your second method of installation is too much for me unfortunately, perhaps I will look for some Youtube tutorial on that topic.

Sorry again. I really appreciate all your hard work and effort.

suaveolent, 2024-02-22:
In the HACS tab, can you see the Hoymiles HMS-XXXXW-T2 integration? 

If yes, you should see a screen something like this:

[image]

Eithere there is a blue download button at the right lower corner or you can click on the three dots on the top right to see if you can install it there.

closed 2024-02-22 by Kikash10

suaveolent, 2024-02-22:
I see, this seems to be the "experimental" feature documented here:
https://experimental.hacs.xyz

Great that it works for you now!

Kikash, 2024-02-22:
Thanks for your help once again!:-)
Everything is experimental for me right now.lol
Do you know why in dashboard sometimes values go to unknown but AC Power gets set to 1,000.18 W? I am assuming it wasn't able to fetch new info at that attempt(I have set my interval to be every 60 seconds) and is using some predetermined values? Did anyone else have experience this situation?

suaveolent, 2024-02-22:
This seems to be related with the following issue and should be fixed with the next release (or just update to main branch in the meantime)

https://github.com/suaveolent/ha-hoymiles-wifi/issues/11