suaveolent/ha-hoymiles-wifi #14: Since 0.1.3 it frequently loses connection to DTU

julianxv, 2024-02-24:
Since the 0.1.3 Update it looses frequently after a few hours the connection the DTU. I can only force a reconnect by restarting the whole Home Assistant System. 

I just started debug logging and will update this issue when i got the first logs. 

Home Assistant core 2024.2.2 on a rapsberry Pi 3B. Connected to a HMS800W-2T

suaveolent, 2024-02-24:
Thanks for reporting. 

This is most liekely related to #11 and #12. 

A temporary fix is already pushed to `main` branch. I will release a new version today or tomorrow.

julianxv, 2024-02-24:
Thanks for the quick reply and fix.

thitcher, 2024-02-24:
Same problem here.
But I don't get the inverter back with a restart of HA.
How can I get the fix?
Tried to download the main via HA, but it don't work, HA only let me download the newest version

Holger2024, 2024-02-24:
For me it even kicked the inverter from the wireless network and I had to reset via hotspot of the inverter. 
Then it worked for a day with a disconnect every couple hours. Since this morning it is back to disconnect every couple minutes (even though I manually added the wait time from the last change in HA).

I have a feeling that the inverter is somehow overloaded - e.g. the ping time randomly varies between 5ms to 150ms every ping.

suaveolent, 2024-02-24:
Newest version local looks good. 
I will push a new release this evening.

suaveolent, 2024-02-24:
`v0.1.4` just released, please reopen issue if problem still persists

Holger2024, 2024-02-26:
It worked fine for 1.5 days, but at 8 am this morning the disconnects started again.

[image]

Is it possible that you assume a complete reboot within 24hours, because I'm constantly operating the inverter from batteries?

suaveolent, 2024-02-26:
That is very curious. Maybe it is a bug in the firmware?

Yes, my inverter reboots at night, since I do not have any batteries. I implemented a "Restart" button, can you check if the restart fixes your problem?

thitcher, 2024-02-26:
Same here, worked fine yesterday, today it won´t connect correct.

[image]

thitcher, 2024-02-26:
> Great, please let me know if this fixes your problem.
> 
> The update process can be quite cumbersome, if you are having problems going via the official ways, you can try to update via the `hoymiles-wifi` CLI tool:
> 
> https://github.com/suaveolent/hoymiles-wifi

Unfortunately, I´m not so good in this things, I think the update can´t be done via HA?
So I have to wait till I´m back@home

thitcher, 2024-02-26:
> I'm working on integrating the update mechanim, but unfortunately this is not yet supported in HA. You can try updating via the Hoymiles Cloud, though: https://global.hoymiles.com/platform/login

Thats what I try, but the Inverter shows Offline, so I can´t do an update.
Somentimes it is online, when I have luck and hit that minute, I try to update it.
Otherwise I have to do it this afternoon vie Smartphone and direct connection.

suaveolent, 2024-02-26:
Yes, that seems like a good idea. Alternatively you can try temporarily disabling the HA integration to see if the inverter bounces back and then try the update again via Hoymiles cloud.

thitcher, 2024-02-26:
I had luck! 
Update to XX.XX.11 was successful.
Now, it seems to be stable:
In the meantime I searched the web and found this article:
[https://mikrowechselrichter.eu/blogs/neuigkeiten/hoymiles-hms-800w-2t-verbindungsprobleme-wlan-server-s-miles-cloud]
Maybe the Problem is this: 

### Was ist die Ursache des Problems?

Hoymiles hat bei der Entwicklung der Firmware - entgegen jedem normalen Vorgehen in der Softwareentwicklung - feste IP-Adressen für DNS-Server hinterlegt.

Was heisst das?

Der Mikrowechselrichter versucht, über das Internet auf die Cloud zuzugreifen. Diese ist unter der Domain dataeu.hoymiles.com erreichbar.

Um die Adresse dataeu.hoymiles.com in eine "maschinenlesbare" IP-Adresse zu übersetzen, braucht es einen sog. DNS-Dienst. Normalerweise übernimmt diese Funktion der eigene Internetanbieter automatisch für jeden Internetanschluss.

Hoymiles hat statt dessen aber die Verwendung eines eigenen DNS-Dienstes direkt in die Geräte einprogrammiert. Die beiden dafür fest hinterlegten IP-Adressen der DNS-Server (165.165.165.165 und 28.31.9.0) erfüllen aber ihre Aufgabe nicht und sind nicht erreichbar.

Deswegen kann der Mikrowechselrichter der Domain dataeu.hoymiles.com keine IP-Adresse zuordnen und somit den eigenen Cloud-Server nicht erreichen.

### What is the cause of the problem?

When developing the firmware - contrary to any normal procedure in software development - Hoymiles stored fixed IP addresses for DNS servers.

What does that mean?

The microinverter tries to access the cloud via the Internet. This can be reached under the domain dataeu.hoymiles.com.

In order to translate the address dataeu.hoymiles.com into a “machine-readable” IP address, a so-called DNS service is required. Normally, your own internet provider automatically takes over this function for every internet connection.

Instead, Hoymiles has programmed the use of its own DNS service directly into the devices. However, the two IP addresses of the DNS servers stored for this purpose (165.165.165.165 and 28.31.9.0) do not fulfill their task and cannot be reached.

Therefore, the microinverter cannot assign an IP address to the domain dataeu.hoymiles.com and therefore cannot reach its own cloud server.

thitcher, 2024-02-27:
The values are rock solid since 7:30AM (start of producing energy).
Seems to work perfect now!
Thanks for the code and great support.

julianxv, 2024-02-28:
It disconnects at night. 

The versions are matching.

I activated debug logging and will post a log tomorrow.

suaveolent, 2024-02-28:
> It disconnects at night.
> 
> The versions are matching.
> 
> I activated debug logging and will post a log tomorrow.

Great! Don’t forget to restart your Home Assistant instance after changing the configuration.

julianxv, 2024-03-01:
Now it's running stable for the last two days.

suaveolent, 2024-03-02:
Got my first disconnect today. And seeing at least one disconnect a day. I'm starting to wonder, if this is because of the inverter sequence counter overflowing (or some internal bug in the DTU).

I'll try to statically set the counter to 1 and see if that improves anything.

suaveolent, 2024-03-03:
After setting the sequence counter to 1 I did not have a single disconnect today. I will keep monitoring this and if successful push a fix.

suaveolent, 2024-03-22:
My connection has been rock solid over the past weeks. I'm beginning to wonder, if this issue was related to the Hoymiles Cloud and not the custom component.

@julianxv @thitcher are you still seeing disconnects?

julianxv, 2024-03-22:
For me it's the same. No connection losses in the past weeks.

suaveolent, 2024-03-22:
Im closing the issue for now.

closed 2024-03-22 by suaveolent