suaveolent/ha-hoymiles-wifi #15: Guide how to use the data for the energy page in HA

ThomWee, 2024-02-26:
Thanks for this extension.

**Describe the bug**
I'm trying to configure the energy tab in my HA making use of the current Wh produced by my panels. But somehow the data displayed are never matching the output. I'm new to this topic, so maybe I'm understanding something wrong.

**To Reproduce**
My HA instance is running as a docker image on my raspberry pi. HA is running as version 2024.1.5 and your extension in version 0.1.4. Language I'm using is German.

I did configure the energy page to display Port 1/2 DC power.
[image]

But the final results are just wrong. I'm having a 800Wh solar installation and can't make 8kWh.
[image]

It would be nice to get some hints how to correct setup HA displaying my data.

This is if I show the extension data on the dashboard.
[image]

suaveolent, 2024-02-26:
Your screenshot looks correct.
The first image shows the total number of kW/h for the time from 22.02.2024 - 26.02.2024 (see upper right corner).

So the 8.8kW/h are not for one day but for the entire week.

ThomWee, 2024-02-27:
@suaveolent thanks for reply. I thought the view in the upper right corner are the actual values from right now, because of the nice animation.
When checking the history for AC daily energy, the board shows me that this value is always 0 or like in my second screenshot Unknown. Do I need to tweak something in the extension?

saxn-paule, 2024-02-27:
It works if you use the daily AC Power:
[image]

[image]

suaveolent, 2024-02-27:
> @suaveolent thanks for reply. I thought the view in the upper right corner are the actual values from right now, because of the nice animation. When checking the history for AC daily energy, the [...]

To get the same animation for live data you can use the following plugin:
https://github.com/flixlix/power-flow-card-plus

Yes, I noticed that the power was 0 or unknown before. Are you running the latest version of the integration (v0.1.4)?

Can you maybe navigate to 
Settings > Devices & Services >  Hoymiles HMS-XXXXW-T2 > Hoymiles HMS-XXXXW-T2 

and see if the sensor "AC power" shows up (entity id should be sensor.hoymiles_hms_xxxxw_t2_ac_power)?

ThomWee, 2024-02-27:
I can see this entity but no data are present, just 0. Can I somehow create logs or some printout helping to debug my problem?
[image]

ThomWee, 2024-02-27:
And yes I'm using 0.1.4

suaveolent, 2024-02-27:
Are you familiar with CLI / python pip? 

You can try running the ´get-real-data-new` request from the underlying library itself. See:

https://github.com/suaveolent/hoymiles-wifi

e.g. 

`hoymiles-wifi --host <IP-Address or Hostname of DTU> get-real-data-new`

ThomWee, 2024-02-27:
> Get-real-data-new Response: 
device_serial_number: "413******"
timestamp: 1709048682
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 189***************
  firmware_version: 1
  voltage: 2396
  frequency: 4996
  active_power: 168
  current: 7
  power_factor: 1000
  temperature: 98
  warning_number: 1
  crc_checksum: 4099
  link_status: 1
  modulation_index_signal: 2097182
}
pv_data {
  serial_number: 189************
  port_number: 1
  voltage: 412
  current: 21
  power: 89
  energy_total: 37325
  energy_daily: 229
  error_code: 50331648
}
pv_data {
  serial_number: 189***********
  port_number: 2
  voltage: 409
  current: 21
  power: 88
  energy_total: 34390
  energy_daily: 216
  error_code: 50331648
}

> Get-version-info Response: 
dtu_hw_version: "H06.02.01"
dtu_sw_version: "V00.03.12"
inverter_hw_version: "H00.04.00"
inverter_sw_version: "V01.00.02"

suaveolent, 2024-02-27:
Thanks, It seems like you have a completely different HW/SW version than I have and the `dtu_power` is missing in your response. 

Have you already upgraded to latest DTU version (V00.01.11 for me)?

This might fix your problem. Otherwise I might also be able to fix it in a future update using the `active_power` of `sgs_data`.

ThomWee, 2024-02-27:
Hmm, I updated my DTU now from SW version 00.03.12 to 00.03.13 ... now I get way less data 

> Get-real-data-new Response: 
device_serial_number: "4133A00323C2"
timestamp: 1709051117
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 18985986852358
  modulation_index_signal: 7929856
}
pv_data {
  serial_number: 18985986852358
  port_number: 1
}
pv_data {
  serial_number: 18985986852358
  port_number: 2
}

suaveolent, 2024-02-27:
Thanks, seems like this info is not provided with your DTU. I will fix it with next version. 

Btw: Which inverter model do you have?

ThomWee, 2024-02-27:
Thanks. I added the active_power sensor by hand now. Let's see tomorrow if there are data send to my HA sensors.

[image]

suaveolent, 2024-02-28:
Pushed fix to main, will be fixed with next release.

closed 2024-02-28 by suaveolent