suaveolent/ha-hoymiles-wifi #17: DTU Wlite-S with HMS-1600-4T

Philipp23021991, 2024-02-28:
Hi,

i used your Integration with HMS-1600-4T and DTU Wlite-S.

All data from the DTU are fine.

These data work from the hoymiles:
Grid frequency
Grid voltage
Port 1 DC Current
Port 1 DC Daily Energy
Port 1 DC Power
Port 1 DC Total Energy
Port 1 DC Voltage
Port 2 DC Current
Port 2 DC Daily Energy
Port 2 DC Power
Port 2 DC Total Energy
Port 2 DC Voltage
Temperature

this data don‘t work:
AC Daily Power
AC Power
Port 3 data
Port 4 data

Is it possible, to add these features to your Integration? Can you make it? Can i help you ?

Thank You!

suaveolent, 2024-02-29:
Yes should be possible.

Can you post the output of the  `get-real-data-new` command from the underlaying `https://github.com/suaveolent/hoymiles-wifi` library? (Please only keep the first 4 digits of the serial numbers)

Philipp23021991, 2024-02-29:
.fr-emoticon.fr-emoticon-img {
                                background-repeat: no-repeat !important;
                                font-size: inherit;
                                height: 1em;
                                width: 1em;
                                min-height: 20px;
                                min-width: 20px;
                                display: inline-block;
                                margin: -0.1em 0.1em 0.1em;
                                line-height: 1;
                                vertical-align: middle;
                            }
                            .fr-emoticon {
                                font-weight: normal;
                                font-family: "Apple Color Emoji","Segoe UI Emoji","NotoColorEmoji","Segoe UI Symbol","Android Emoji","EmojiSymbols";
                                display: inline;
                                line-height: 0;
                            }
                        Hey,
thank you for your answer. 
Here are the outputs:
 Get-real-data-new Response:device_serial_number: &quot;4133xxxxxxxx&quot;timestamp: 1709206215active_power: 1firmware_version: 1sgs_data {  serial_number: 1912xxxxxxxxxx  firmware_version: 1  voltage: 2422  frequency: 4999  active_power: 6924  reactive_power: 5  current: 285  power_factor: 999  temperature: 176  warning_number: 44  link_status: 1  modulation_index_signal: 100}pv_data {  serial_number: 1912xxxxxxxxxx  port_number: 1  voltage: 334  current: 397  power: 1327  energy_total: 538  energy_daily: 158  error_code: 50331648}pv_data {  serial_number: 1912xxxxxxxxxx  port_number: 2  voltage: 334  current: 403  power: 1349  energy_total: 499  energy_daily: 166  error_code: 50331648}pv_data {  serial_number: 1912xxxxxxxxxx  port_number: 3  voltage: 349  current: 489  power: 1709  energy_total: 384  energy_daily: 177  error_code: 50331648}pv_data {  serial_number: 1912xxxxxxxxxx  port_number: 4  voltage: 336  current: 864  power: 2902  energy_total: 440  energy_daily: 242  error_code: 50331648}
Get-real-data-hms Response:dtu_sn: &quot;4133xxxxxxxx&quot;time: 1709206438device_nub: 1package_nub: 1inverter_state {  inv_id: 1912xxxxxxxxxx  port_id: 1  grid_voltage: 2422  grid_freq: 4999  pv_current_power: 7007  unknown1: 289  unknown2: 999  temperature: 181  unknown3: 44  unknown4: 1  bit_field: 100}port_state {  pv_sn: 1912xxxxxxxxxx  pv_port: 1  pv_vol: 334  pv_cur: 413  pv_power: 1383  pv_energy_total: 546  pv_daily_yield: 166  bitfield: 50331648}port_state {  pv_sn: 1912xxxxxxxxxx  pv_port: 2  pv_vol: 335  pv_cur: 419  pv_power: 1405  pv_energy_total: 507  pv_daily_yield: 174  bitfield: 50331648}port_state {  pv_sn: 1912xxxxxxxxxx  pv_port: 3  pv_vol: 349  pv_cur: 490  pv_power: 1716  pv_energy_total: 394  pv_daily_yield: 187  bitfield: 50331648}port_state {  pv_sn: 1912xxxxxxxxxx  pv_port: 4  pv_vol: 337  pv_cur: 851  pv_power: 2871  pv_energy_total: 458  pv_daily_yield: 260  bitfield: 50331648}
Get-real-data Response:dtu_sn: &quot;4133xxxxxxxx&quot;timestamp: 1709206521device_number: 1package_number: 1rp_data {  rp_sn: 1912xxxxxxxxxx  rp_signal: 1  rp_channel: 2425  rp_link_nub: 4998  rp_link_status: 7037}
App-information-data Response:dtu_serial_number: &quot;4133xxxxxxxx&quot;timestamp: 1709206622device_number: 1pv_number: 4package_number: 1channel: 1dtu_info {  dtu_sw_version: 781  dtu_hw_version: 25089  dtu_step_time: 5  dtu_rf_hw_version: 1282  access_model: 1  communication_time: 1709206571  signal_strength: 82  wifi_version: &quot;2.3.0.0-dev&quot;  dtu_rule_id: 18  sub1g_frequency_band: 2  sub1g_channel_total_number: 21  sub1g_channel_number: 33  sub1g_rp: 20  sub1g_channel_total: 10}pv_info {  pv_serial_number: 1912xxxxxxxxxx  pv_sw_version: 10027  pv_hw_part_number: 270680326  pv_hw_version: 266  pv_grid_profile_code: 768  pv_grid_profile: 8192  pv_rf_hw_version: 84017154  pv_rf_sw_version: 5}
Get-information-data Response:dtu_sn: &quot;4133xxxxxxxx&quot;time: 1709206708device_nub: 1pv_nub: 4package_nub: 1channel: 1mDtuInfo {  dtu_sw: 781  dtu_hw: 25089  dtu_step_time: 5  dtu_rf_hw: 1282  access_model: 1  grid_type: 18  surplus_power_c: 2  zero_export_control: 21  phase_balance_switch: 33  tolerance_between_phases: 20}mpvInfo {  pv_sn: 1912xxxxxxxxxx  pv_sw: 10027  pv_hw_pn: 270680326  pv_hw: 266  pv_gpf_code: 768  pv_gpf: 8192  pv_rf_hw: 84017154  pv_rf_sw: 5}
Get-version-info Response:dtu_hw_version: &quot;H06.02.01&quot;dtu_sw_version: &quot;V00.03.13&quot;inverter_hw_version: &quot;H00.04.10&quot;inverter_sw_version: &quot;V01.00.27&quot;

-----urspr&uuml;ngliche Nachricht-----

suaveolent, 2024-02-29:
Thanks. Your DTU reports quite a lot data in comparison to the HMS-XXXW-2T series. 
Reading the data is not a big deal.
However, It will take me a while to get a dyanmic process working (i.e. only show the actual number of sensors instead of alaways trying to display 4 ports)

Philipp23021991, 2024-03-03:
Thanks to You!
If you have some questions please ask me.

spike-23, 2024-03-04:
Hi @suaveolent, thanks for your work. I also just installed your integration in conjunction with my HMS-1600-4T. I would highly appreciate a working solution for it! Would be glad to beta test or provide feedback.

LuucA80, 2024-03-04:
Hi @suaveolent also thanks from my side for your work. I have also installed your ingetration.
In my setup I use 3 HMS-2000-4T. I would highly appreciate a working solution for it! If you need something from my side or test something let me know.

suaveolent, 2024-03-04:
Yes, I'm already working on it.
Can you maybe too post the output of the `get-real-data-new` and `app-information-data` commands from the underlaying https://github.com/suaveolent/hoymiles-wifi library?

Please snip the serial numbers to the first 4 digits.

T4s-H4g, 2024-03-04:
Hi @suaveolent,
I have both here. The HMS-1600-4T with the DTU Wlite-S and a HMS-800W-2T. If I can help with tests or outputs from the HMS let me know.
Best,
Thomas

suaveolent, 2024-03-04:
@T4s-H4g thanks, I should have these devices already covered. 

What I am interested is a setup where multiple inverters are behind a single DTU.

LuucA80, 2024-03-05:
Hi @suaveolent,

Here the output of the two commands

get-real-data-new
`Get-real-data-new Response:
device_serial_number: "10FDxxxxxxxx"
timestamp: 1709623841
active_power: 2
firmware_version: 1
sgs_data {
  serial_number: 1912xxxxxxxxxxx
  firmware_version: 1
  voltage: 2267
  frequency: 5003
  active_power: 113
  reactive_power: -4847
  current: 4
  power_factor: 1
  temperature: 56
  warning_number: 9
  crc_checksum: 4242
  link_status: 1
  modulation_index_signal: 670763239
}
sgs_data {
  serial_number: 1912xxxxxxxxxx
  firmware_version: 1
  voltage: 2274
  frequency: 5003
  active_power: 113
  reactive_power: -4346
  current: 5
  power_factor: 1
  temperature: 55
  warning_number: 9
  crc_checksum: 20093
  link_status: 1
  modulation_index_signal: 672663795
}
sgs_data {
  serial_number: 1912xxxxxxxxxx
  firmware_version: 1
  voltage: 2270
  frequency: 5003
  active_power: 122
  reactive_power: -1
  current: 5
  power_factor: 1
  temperature: 56
  warning_number: 9
  crc_checksum: 27098
  link_status: 1
  modulation_index_signal: 675023079
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 1
  voltage: 280
  current: 11
  power: 31
  energy_total: 414577
  energy_daily: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 2
  voltage: 273
  current: 10
  power: 29
  energy_total: 405705
  energy_daily: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 3
  voltage: 279
  current: 10
  power: 28
  energy_total: 407699
  energy_daily: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 4
  voltage: 278
  current: 10
  power: 29
  energy_total: 401230
  energy_daily: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 1
  voltage: 281
  current: 10
  power: 29
  energy_total: 227832
  energy_daily: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 2
  voltage: 279
  current: 10
  power: 29
  energy_total: 229828
  energy_daily: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1912xxxxxxxxxx
  port_number: 3
  voltage: 276
  current: 10
  power: 29
  energy_total: 230310
  energy_daily: 1
  error_code: 50331648
}`

app-information-data
`App-information-data Response:
dtu_serial_number: "10FDxxxxxxxxx"
timestamp: 1709624127
device_number: 3
pv_number: 12
package_number: 1
channel: 1
dtu_info {
  dtu_sw_version: 526
  dtu_hw_version: 38401
  dtu_rf_hw_version: 131074
  dtu_rf_sw_version: 5
  access_model: 1
  communication_time: 1709624120
  signal_strength: 88
  wifi_version: "2.3.0.0-deX"
  dtu_rule_id: 21
  sub1g_frequency_band: 2
  sub1g_channel_total_number: 21
  sub1g_channel_number: 33
  sub1g_rp: 20
  sub1g_channel_total: 10
}
pv_info {
  pv_serial_number: 1912xxxxxxxxxx
  pv_sw_version: 10014
  pv_hw_part_number: 269644033
  pv_hw_version: 256
  pv_grid_profile_code: 2560
  pv_grid_profile: 8193
}
pv_info {
  pv_serial_number: 1912xxxxxxxxxx
  pv_sw_version: 10014
  pv_hw_part_number: 269644033
  pv_hw_version: 256
  pv_grid_profile_code: 2560
  pv_grid_profile: 8193
}
pv_info {
  pv_serial_number: 1912xxxxxxxxxx
  pv_sw_version: 10014
  pv_hw_part_number: 269644033
  pv_hw_version: 256
  pv_grid_profile_code: 2560
  pv_grid_profile: 8193
}`

suaveolent, 2024-03-05:
Thanks, that helps a lot!

suaveolent, 2024-03-14:
@Philipp23021991 @spike-23 @T4s-H4g @LuucA80

I have just pushed a test version to the [multi_inverter](https://github.com/suaveolent/ha-hoymiles-wifi/tree/multi_inverter) branch.
If you want to beta test you can upgrade your custom component to that branch.

**Note**: There are many changes in the custom component. If you experience any problems, please report them first here. Then, it might be necessary to set-up the config once again to ensure that everything is set-up correctly. 
Also bear in mind, that the turn-on and turn-off buttons might only work for one inverter yet.

LuucA80, 2024-03-14:
Thanks. I just installed this version in a test environment. I will let you know my experience.

suaveolent, 2024-03-15:
Thank

> Thanks. I just installed this version in a test environment. I will let you know my experience.
> 
> @suaveolent. First of all thanks for the good work so far. Here two points I see.
> 
> 1. My setup is a DTU-S pro with 3 HMS-2000-4T. After installation I see the DTU and 3 inverters presented as model HMS-1600-4T
> 2. On the first inverter I see the 4 channels. On the second inverter I see 3 channels and on the third inverter I see none channels.

Thanks, can you maybe post the config entry for your setup. You can find it in the `.storage/core.config_entries`  file. (Please again snip the serial numbers to the first 4 digits)

What I'm interested in is especially the `"ports"` part. 

Afterwards, can you try setting the config up again and see if anything changes?

LuucA80, 2024-03-15:
Here you are.

     {
        "entry_id": "2bf707795c0fad8f68f7e8febd834d88",
        "version": 2,
        "minor_version": 1,
        "domain": "hoymiles_wifi",
        "title": "192.168..x.x",
        "data": {
          "host": "192.168.x.x",
          "sensor_prefix": "",
          "update_interval": 35,
          "dtu_serial_number": "10FD",
          "inverters": [
            "1164",
            "1164",
            "1164"
          ],
          "ports": [
            {
              "inverter_serial_number": "1164",
              "port_number": 1
            },
            {
              "inverter_serial_number": "1164",
              "port_number": 2
            },
            {
              "inverter_serial_number": "1164",
              "port_number": 3
            },
            {
              "inverter_serial_number": "1164",
              "port_number": 4
            },
            {
              "inverter_serial_number": "1164",
              "port_number": 1
            },
            {
              "inverter_serial_number": "1164",
              "port_number": 2
            },
            {
              "inverter_serial_number": "1164",
              "port_number": 3
            }
          ]
        },
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "unique_id": null,
        "disabled_by": null
      }

suaveolent, 2024-03-15:
> Here you are.
> 
> ```
>  {
>     "entry_id": "2bf707795c0fad8f68f7e8febd834d88",
>     "version": 2,
>     "minor_version": 1,
>     "domain": "hoymiles_wifi",
>     "title": "192.168..x.x",
>     "data": {
>       "host": "192.168.x.x",
...
> ```

Thanks, that confirms my assumption from above, that the data is not retrieved via the `hoymiles-wifi` lib. Now I need to figure out why that is.

Philipp23021991, 2024-03-15:
Hi @suaveolent,

I upgraded and it works :)
[image]
Thank You =)

I will test it some days and give you a feedback.

LuucA80, 2024-03-15:
> @LuucA80 although this should not really matter, since I implementeted the CRC check, I'd still like to verify if there is an issue with e.g. the response from the DTU.
> 
> Can you maybe change: in the `dtu.py` file of the `hoymiles-wifi` library the following lines, and run the `get-real-data-new` command once again.
> 
> **Insert** after the line `crc16_target, read_length = struct.unpack(">HH", buf[6:10])` insert `print(f"Read length: {read_length}")`
> 
> **Change:** ` buf = await asyncio.wait_for(reader.read(1024), timeout=5)` to `buf = await asyncio.wait_for(reader.read(2048), timeout=5)`
> 
> `response_as_bytes = buf[10:read_length]` to `response_as_bytes = buf[10:2048]`
> 
> Please also reply with the `Read length` output.

Sure I can. The output of 'Read length' is 425

suaveolent, 2024-03-15:
> Sure I can. The output of 'Read length' is 425

Thanks, so that's not the issue.

Is there maybe a software update available for the DTU and aan you access all live data in the hoymiles app? 
If not, it might be a bug in the DTU itself.

Otherwise, we'll have to look deeper and maybe even capture traffic between the app and the inverter to evaluate what the app is requesting.

LuucA80, 2024-03-15:
I don't know if there is a software update. When I logged in in the app there is no message or something like that.
The firmware I use is:
[image]

In the app (online and app on telephone) I see everything.

T4s-H4g, 2024-03-15:
Hi @suaveolent,

I upgraded and it works :-) Thanks a lot for your work!

- the 4 ports of my HMS-1600-4T are displayed correctly.
- the Powerlimit is set in the Hoymiles app to 37% and in HA it is 0.
- the AC daily energy for my DTU Wlite-S is displayed as 'unknown'. The Debug log shows: `2024-03-15 17:11:08.281 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for AC daily energy to avoid resetting total_increasing counter`
- for the DTU-HMS-W the AC daily energy is in HA 69Wh and in the App 148Wh

suaveolent, 2024-03-15:
> Hi @suaveolent,
> 
> I upgraded and it works :-) Thanks a lot for your work!

Thanks for the feedback. 

> * the Powerlimit is set in the Hoymiles app to 37% and in HA it is displayed 0.

Can you maybe provide the output of the `get-config` command from the underlaying https://github.com/suaveolent/hoymiles-wifi library? Please only keep the first 4 digits of the serial numbers.

> * the AC daily energy for my DTU Wlite-S is displayed as 'unknown'. The Debug log shows: `2024-03-15 17:11:08.281 DEBUG (MainThread) [custom_components.hoymiles_wifi.sensor] Returning last known value instead of 0.0 for AC daily energy to avoid resetting total_increasing counter`
Afaik the DTU Wlite-S does not report the AC daily energy. That why it is 'unknown'. It only works for the HMS W series.

> for the DTU-HMS-W the AC daily energy is in HA 69Wh and in the App 148Wh

Could it be that the App includes the daily energy from both inverters?

T4s-H4g, 2024-03-16:
Here is the output of the `get-config`:
```
Get-config Response: 
request_offset: 3600
request_time: 1710589508
netmode_select: 1
server_send_time: 15
wifi_rssi: 62
serverport: 10081
apn_set: "CMNET"
meter_kind: "2.3.0.0-dev"
meter_interface: "2.3.0.0-dev"
wifi_ssid: "XXX"
wifi_password: "XXX"
server_domain_name: "dataeu.hoymiles.com"
dtu_sn: "4133xxx"
access_model: 1
mac_0: xx
mac_1: xx
mac_2: xxx
mac_3: x
apn_name: "NONE"
apn_password: "NONE"
sub1g_work_channel: 9
cable_dns_0: 1
cable_dns_1: 1
cable_dns_2: 1
cable_dns_3: 1
wifi_ip_addr_0: xxx
wifi_ip_addr_1: xxx
wifi_ip_addr_2: xxx
wifi_ip_addr_3: xxx
mac_4: xx
mac_5: x
wifi_mac_0: x
wifi_mac_1: xx
wifi_mac_2: xxx
wifi_mac_3: xxx
wifi_mac_4: xx
wifi_mac_5: x
dtu_ap_ssid: "XXX"
dtu_ap_pass: "XXX"
```

> Afaik the DTU Wlite-S does not report the AC daily energy. That why it is 'unknown'. It only works for the HMS W series.

OK. Than I have to calculate it.

> Could it be that the App includes the daily energy from both inverters?

No, they are seperated systems in the App.

Today I have 521Wh in the App and 539Wh in HA. Could it be because of the fact that the App is not so often updated?
What I see today in the App is that the updates of  the HMS-800W-2T is around 1,5 hours behind of the updates from my HMS-1600-4T. The HMS-1600-4T is updated every 15 minutes.

suaveolent, 2024-03-18:
> I don't know if there is a software update. When I logged in in the app there is no message or something like that. The firmware I use is: [...]

Thanks, I was referring to the live data (see p. 16):

https://assets.mikrowechselrichter.eu/hoymiles-dtu-wlite-s/en/user-manual-hoymiles-dtu-lite-s.pdf

suaveolent, 2024-03-18:
> Here is the output of the `get-config`:
>
> ```
> Get-config Response:
> request_offset: 3600
> request_time: 1710589508
> netmode_select: 1
> server_send_time: 15
> wifi_rssi: 62
> serverport: 10081
> apn_set: [...]

Thanks, that explains why the power limit is 0. This is apparently a feature of the HMS with integrated DTU series. 

Can you try all the different commands provided by the `hoymiles-wifi` library, to see if there is anything related to that power limit? Otherwise, we'll have too capture the network traffic to find that out.

LuucA80, 2024-03-18:
> > I don't know if there is a software update. When I logged in in the app there is no message or something like that. The firmware I use is: 
> > In the app (online and app on telephone) I see everything.
> 
> Thanks, I was referring to the real time data (see p. 16):
> 
> https://assets.mikrowechselrichter.eu/hoymiles-dtu-wlite-s/en/user-manual-hoymiles-dtu-lite-s.pdf

Hi @suaveolent,

Unfortunately I have a end-user account with in Hoymiles. So I don't see the tab O&M.
What I see is that the data is been refreshed every 15 minutes.

I also use a DTU-Pro and not a DTU-Lite. Don't know if there is a difference?

suaveolent, 2024-03-18:
> Hi @suaveolent,
>
> Unfortunately I have a end-user account with in Hoymiles. So I don't see the tab O&M. What I see is that the data is been refreshed every 15 minutes.
>
> I also use a DTU-Pro and not a [...]

@LuucA80, I see. Can't really tell if there is a difference between these devices, since I have neither of those. 
Can maybe too try all the different commands provided by the `hoymiles-wifi` lib (e.g. especially the `get-real-data` command), to see if the remaining ports are reported with a different command?

T4s-H4g, 2024-03-20:
> > Here is the output of the `get-config`:
> > ...
> > > Afaik the DTU Wlite-S does not report the AC daily energy. That why it is 'unknown'. It only works for the HMS W series.
> > 
> > 
> > OK. Than I have to calculate it.
> > > Could it be that the App includes the daily energy from both inverters?
> > 
> > 
> > No, they are seperated systems in the App.
> > Today I have 521Wh in the App and 539Wh in HA. Could it be because of the fact that the App is not so often updated? What I see today in the App is that the updates of the HMS-800W-2T is around 1,5 hours behind of the updates from my HMS-1600-4T. The HMS-1600-4T is updated every 15 minutes.
> 
> Thanks, that explains why the power limit is 0. This is apparently a feature of the HMS with integrated DTU series.
> 
> Can you try all the different commands provided by the `hoymiles-wifi` library, to see if there is anything related to that power limit? Otherwise, we'll have too capture the network traffic to find that out.

I tried the hoymiles-wifi v0.1.7 and there is the value for the power limit.
``` 
Get-config Response: 
request_offset: 3600
request_time: 1710955442
limit_power_mypower: 400
netmode_select: 1
server_send_time: 15
wifi_rssi: 50
serverport: 10081
apn_set: "CMNET"
meter_kind: "2.3.0.0-dev"
meter_interface: "2.3.0.0-dev"
wifi_ssid: "xxx"
wifi_password: "xxx"
server_domain_name: "dataeu.hoymiles.com"
dtu_sn: "4133xxx"
access_model: 1
mac_0: xx
mac_1: xx
mac_2: xxx
mac_3: x
apn_name: "NONE"
apn_password: "NONE"
sub1g_work_channel: 1
cable_dns_0: 1
cable_dns_1: 1
cable_dns_2: 1
cable_dns_3: 1
wifi_ip_addr_0: xxx
wifi_ip_addr_1: xxx
wifi_ip_addr_2: xxx
wifi_ip_addr_3: xx
mac_4: xx
mac_5: x
wifi_mac_0: x
wifi_mac_1: xx
wifi_mac_2: xxx
wifi_mac_3: xxx
wifi_mac_4: xx
wifi_mac_5: x
dtu_ap_ssid: "DTUL-xxx"
dtu_ap_pass: "xxx"
```

suaveolent, 2024-03-21:
> I tried the hoymiles-wifi v0.1.7 and there is the value for the power limit.
>
> ```
> Get-config Response:
> request_offset: 3600
> request_time: 1710955442
> limit_power_mypower: 400
> netmode_select: 1 [...]

That's interesting. So it should work in theory. Did you do a fresh setup of the integration or did you upgrade your existing config?

Gazonga83, 2024-04-03:
Hallo zusammen, ich habe eine Open DTU für HMS1600T,
nun scheint es so als ob entweder das Open DTU Daten bekommt oder der Smiles Stick.

Ich hätte gerne das ich beide nutzen kann, also die Smiles cloud und das Open DTU um den Wechselrichter drosseln zu können. 

Habt ihr eine Idee?

Besten Dank

closed 2024-05-18 by github-actions[bot]

justinh998, 2025-01-06:
> That's interesting. So it should work in theory. Did you do a fresh setup of the integration or did you upgrade your existing config?

Hello, I have a similar problem to t4s.

But I don't even have the power limit entity in home assistant.

I reinstalled the integration about 2 weeks ago (i.e. removed in ha config , uninstalled in home , restarted ha , installed in hacs . Restarted ha, added integration again.

But that didn't change anything.