suaveolent/ha-hoymiles-wifi #21: Change Power Limit for each Port individually

danielr0815, 2024-03-12:
I don't know if the micro inverter is supporting it... so just a question: is it possible to control the power limit individually for each port?

I need this for limiting the overall power dynamically.... so if some modules are shaded but others bring full power, I want to have the combined power of all modules set to for example to 500W. (we have 600W limit in germany at the moment)

So I'm planning a script, which reduces individual modules so that all together are limited to the global limit. So let's say if the global limit is 500W and each module have 400W peek, at the moment we can only limit each module to 250W. 

But if I know, that one panel is directed to east, the other one to west, the script can set the limit for one panel to 100W because the shaded module can not deliver more anyway, so it's not a real limit. But now the script can configure the other module to 400W limit. Both limits together are not higher than the overall limit, but the module which get's sun at the moment can use it's max power and the shaded module is limited but not really influenced. The script should detect the optimum distribution of module limits itself (based on the current power values.

The script should also work by using more than one micro inverter... so we could combine 3 inverters with together 6 modules. In the future in germany we can use max 2000W peek solar power, but all solar modules together should be limited to 800W at any time. I don't know, if this automation could be added as well to your integration, but as long as we have at least the possibility to set the max for each port individually, everybody can create an automation for that.

saxn-paule, 2024-03-12:
That won't work. Internally the HMS-800 has two separate MPPTs that limit at ~430W even if there is more input power.

I have four panels with 1670Wp in sum. 835Wp per input.

[image]

suaveolent, 2024-03-12:
Unfortunately the inverter does not supporting the power limit per port, only for the entire device. 

But your use case might still be possible: it should not really matter which module produces more. By setting the power limit of the inverter you can always hard limit it to e.g. 600 W. 

In case you use multiple inverters it might get a bit trickier: you would have to dynamically adjust the output power of the inverters based on the reported values.

danielr0815, 2024-03-12:
> Unfortunately the inverter does not support setting the power limit per port, only for the entire device.
> 

Ok, with openDTU I thought it is possible to set individual limits for each string, that's why I asked... but perhaps this is a wrong understanding from my side.

> But your use case might still be possible: it should not really matter which module produces more. By setting the power limit of the inverter you can always hard limit it to e.g. 600 W.
> 

At the moment we can set the limit for a 800W inverter to 75% if we want to reach 600W limit ... so for string 1 the max 400W are limited to 300W and for string 2 the same right? So if module 1 is shaded and is generating only 50W, module 2 is still limited to 300W and both modules are creating together only 350W.  But normally 450 W are possible as well without a limit and without violating the global 600W limit... it's just an software issue, but perhaps this is  hoymiles firmware issue.

> In case you use multiple inverters it might get a bit trickier: you would have to dynamically adjust the output power of the inverters based on the reported values.

Yes exactly, but if I can't set the limit individually for each string, than this is not so perfectly possible. I can implement a regulator, which increments the limits, as long as the global power is less than the threshold but I guess I have always this 30s dead time and I could violate the max value in that time, so this is not a valid way. Whtever we configure, at any time the sum of the individual limits should never exceed the global limit.

If I would be able to set the limit for each string individually, than I could split the global max value (600W) to all strings so that all together will not exceed the global maximum.

Example for 2 800W Inverter with 2 strings per inverter:

So the module 1 which is producing max 200W can be dynamical limited to 200W
Module 2 (current max 300W - dyn. limit 300W)
Module 3 (current max 50W - dyn. limit 50W) 
Module 4 (current max 60W - dyn. limit 50W)

All together are still limited to 600W,  if sun/shadow is changing we can only generate less power, never more than the maximum... so we fulfill the regulations. 

At the moment I can only set the limit statically for all to 
600W/(2x800W) => 37,5% to limit the complete system to 600W max. ... this is max power of 150W for each string.
The overall power in my example would be 150W + 150W + 50W + 60W = 410W

So if it's really not possible to set the limit for each string individually, we could at least set the limit for inverter 1 (module 1+2) dynamically to 250W (for each string) and the limit for inverter2 to 50W:

200W + 250W + 50W + 50W = 600W

Even if the sun is changing and homeassistant dynamic regulation is crashing, it will never generate more than 600W => regulation fulfilled.

danielr0815, 2024-03-12:
> That won't work. Internally the HMS-800 has two separate MPPTs that limit at ~430W even if there is more input power.
> 
> I have four panels with 1670Wp in sum. 800Wp + 870Wp at input 1 / 2.

You connect 2 modules in parallel right? Do you have some problems with this setup, because you have always one module which is generating full power and another one idle in parallel.

My idea was to have 2 inverters with 4 individual strings and limiting to 600/800W by SW.

danielr0815, 2024-03-12:
> Only with bigger inverters you need to set the limit once. The HMS-1600 has four MPPT with a max output of ~430W each. So it doesn't matter which panel has sun or not. The combined output will never be more than 800W (if configured) and one single output never > ~430W

My understanding of the limit is, that it is not  limiting the overall inverter power. It's always limiting the individual string. So if 2 strings of an HMS-1600 have max power, they are still limited, independent on the generated power of string 3 and 4.

saxn-paule, 2024-03-12:
> At the moment we can set the limit for a 800W inverter to 75% if we want to reach 600W limit ... so for string 1 the max 400W are limited to 300W and for string 2 the same right? 

No. You limit the output to 600W. The input could be 400W + 200W nevertheless.

>You connect 2 modules in parallel right? Do you have some problems with this setup, because you have always one module which is generating full power and another one idle in parallel.

No problems as long as you use exactly the same modules in parallel. I use south + easy and east + west direction.

> My understanding of the limit is, that it is not limiting the overall inverter power. It's always limiting the individual string.
Your understanding seems to be wrong. ;)

closed 2024-03-13 by suaveolent