suaveolent/ha-hoymiles-wifi #23: Wrong naming (or am I wrong?) HMS-XXXXW-T2 instead of HMS-XXXXW-2T

Chaosflo, 2024-03-25:
I noticed that the naming is inconsistent.
In GitHub everywhere at least what I could find was HMS-XXXXW-2T but in home assistant, hacs its HMS-XXXXW-T2.
Which I think atleast is wrong (the T and 2 are reversed)

Just a small little thing for my eye ^^

Chaosflo, 2024-03-25:
Here 
https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/hacs.json#L2C12-L2C33

https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/custom_components/hoymiles_wifi/translations/en.json#L5

https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/custom_components/hoymiles_wifi/translations/de.json#L5

https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/custom_components/hoymiles_wifi/strings.json#L5

https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/custom_components/hoymiles_wifi/entity.py#L32

https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/custom_components/hoymiles_wifi/entity.py#L33

https://github.com/suaveolent/ha-hoymiles-wifi/blob/54dd8454785725454d4eb95bdbc5c5d81e6438a5/custom_components/hoymiles_wifi/const.py#L3

I hope I found every occurences =)

suaveolent, 2024-03-25:
Thank you so much for your input.
I noticed that around 2 weeks ago. I am currently working on supporting more models (incl. DTU-Wlite).
There will be a new release shortly which not only includes the reanming from T2 to 2T but also adds support for detecting the correct model.

Chaosflo, 2024-03-25:
Fine! 
Just a notice, if you rename it, be sure to post a breaking change, as the entities will be renamed too! 

[image]
This could then break much ^^
Just so that you have it on the table

suaveolent, 2024-03-25:
> Fine! Just a notice, if you rename it, be sure to post a breaking change, as the entities will be renamed too! [...]

Thanks, already testing that feature. For updating, it seems that even though the device name changes, the entity name stays the same. Only if you do a fresh update, the entity id name seems to change. 

If you are feeling lucky or want to beta test, you can try manually updating to the `multi_inverter` branch. 
You should see two new devices, with the new names. 
The old devices will still be there, but you can manually delete them. (It was too much of a hassle to implement a correct migration strategy ;))

suaveolent, 2024-03-29:
Closing with release of v0.2.0

closed 2024-03-29 by suaveolent