suaveolent/ha-hoymiles-wifi #29: no ha entities/devices are generated after updating to v0.2.1

KeksKartoffel, 2024-04-11:
Hi,
since I updated from version v0.1.4 to version v0.2.1, the data output no longer works. Yes, I have read the Breaking Changes from version v0.2.0 and also have the corresponding HA version in use. However, during the migration, there was a "Migration Error," possibly because the DTU was not reachable (which is not uncommon with Hoymiles).

As a result, I deleted and re-added the connection to Hoymiles from HA. This works so far, but no devices and entities are being displayed anymore.
[image]

Debug Log: 
`
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'translation_key'
2024-04-11 09:44:30.752 ERROR (MainThread) [homeassistant.components.number] Error while setting up hoymiles_wifi platform for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 352, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 533, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 642, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'translation_key'
2024-04-11 09:44:30.760 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up hoymiles_wifi platform for binary_sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 352, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 533, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 642, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'translation_key'
2024-04-11 09:44:30.768 ERROR (MainThread) [homeassistant.components.button] Error while setting up hoymiles_wifi platform for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 352, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 533, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 642, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'translation_key'
`

Can you help here? Is there maybe a bug?

suaveolent, 2024-04-12:
That is exactly the kind of error I would expect from an old version of Home Assistant. 

Are you sure you upgraded your home assistant version correctly and did you restart your instance afterwards?

KeksKartoffel, 2024-04-12:
maybe that is the problem. you're right - i'm currently on version 2024.2.**3**: (i thought, that i updated already to 2024.3) I'll try it with the newest version 2024.4 - but i think this report has been resolved.

suaveolent, 2024-04-12:
Great, let me know if that solves the issue.

KeksKartoffel, 2024-04-13:
yes, it works on 2024.4 - thanks for the quick reply :)

closed 2024-04-13 by KeksKartoffel