suaveolent/ha-hoymiles-wifi #36: Can't connect to host

LarsDBAS, 2024-05-06:
I'm running HA as a container and constantly run into the following error when trying to connect to my HMS-800W-2T.

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:507
Integration: Sensor (documentation, issues)
First occurred: 4. Mai 2024 um 00:04:57 (1 occurrences)
Last logged: 4. Mai 2024 um 00:04:57

Error adding entities for domain sensor with platform hoymiles_wifi
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 616, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'translation_key'

suaveolent, 2024-05-06:
Please update your Home Assistant version. The error indicates that you are running an old version of Home Assistant.

suaveolent, 2024-05-06:
Either the ip address or the host name (dns name)

suaveolent, 2024-05-06:
Is there anything in the logs which might explain the error?

ShaneFK95111, 2024-05-07:
What do I have to use for host? Cant connect…:-(

suaveolent, 2024-05-08:
> What do I have to use for host? Cant connect…:-(

Either the ip address or the host name (dns name)...

ShaneFK95111, 2024-05-08:
> > What do I have to use for host? Cant connect…:-(
> 
> Either the ip address or the host name (dns name)...

Yeah thanks, Bit which device IP? Of hoymiles converter?

suaveolent, 2024-05-08:
> > > What do I have to use for host? Cant connect…:-(
> > 
> > 
> > Either the ip address or the host name (dns name)...
> 
> Yeah thanks, Bit which device IP? Of hoymiles converter?

Yes.

maartentmm, 2024-05-14:
Im up to date and can't connect to host

Core
2024.5.3
Supervisor
2024.05.1
Operating System
12.3
Frontend
20240501.1

suaveolent, 2024-05-14:
Is there anything in the log? 

Can you ping the inverter from your home assistant instance?

maartentmm, 2024-05-14:
> Is there anything in the log?
> 
> Can you ping the inverter from your home assistant instance?

Yes i get pings without timeout.

suaveolent, 2024-05-14:
Can you maybe enable debug logging for the hoymiles-wifi integration and check if there is anything in the logs which might explain the issue?

maartentmm, 2024-05-15:
> Can you maybe enable debug logging for the hoymiles-wifi integration and check if there is anything in the logs which might explain the issue?

its working now with the ethernet disconnected

closed 2024-06-03 by suaveolent