suaveolent/ha-hoymiles-wifi #37: Update 0.2.2 causes energy dashboard to break

N3rdix, 2024-05-10:
**Describe the bug**
Although it is mentioned in the code that the new workaround for #32 should handle zero values correctly it seems to break something.
I have updated to 0.2.2 yesterday and now the total energy sensors jump to zero at night and jump back to the original value when the DTU starts at morning. This of course leads to incorrect solar production values as in my example.

**To Reproduce**
Steps to reproduce the behavior:
1. Update to 0.2.2
2. Add the total energy sensors to the energy dashboard
3. See error

**Expected behavior**
The total energy sensors should not reset to zero at all.

**Screenshots**
[image]
[image]
[image]

**Additional context**
Temporary workaround can be to delete the outlier values every day, so at least nothing will break hard 😄

suaveolent, 2024-05-10:
Thanks, you are right. Will provide a fix later today.

suaveolent, 2024-05-10:
One more comment: you should not use total energy for the energy dashboard. 

If you restart your home assistant instance and the inverter is not available it might return 0 anyway.

Instead use daily energy. It works just as intented even if the sensor is set to 0.

N3rdix, 2024-05-10:
> One more comment: you should not use total energy for the energy dashboard.
> 
> If you restart your home assistant instance and the inverter is not available it might return 0 anyway.
> 
> Instead use daily energy. It works just as intented even if the sensor is set to 0.

Thanks for that hint, will look into this although I did not have any trouble yet (luckily). Historically I've added total sensors the past years, but according to the HA docs your advice might be the best choice 👍

suaveolent, 2024-05-10:
`v0.2.3` should fix the problem.

closed 2024-05-10 by suaveolent

N3rdix, 2024-05-10:
Thanks for the quick resolution, awesome work 👏

suaveolent, 2024-05-10:
You‘re welcome. With the fix you should be able to keep using the total sensor.

commit 0d1dacc2 (2024-05-10): Fixes #37