suaveolent/ha-hoymiles-wifi #38: Platform not found for

elafontaine, 2024-05-15:
**Describe the bug**
Get greated with this error while trying to install on my home assistant ; 

```
Logger: homeassistant.setup
Source: setup.py:403
First occurred: 3:10:33 PM (4 occurrences)
Last logged: 3:10:33 PM

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 834, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 851, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/hoymiles_wifi/sensor.py", line 43, in <module>
    from .entity import HoymilesCoordinatorEntity, HoymilesEntityDescription
  File "/config/custom_components/hoymiles_wifi/entity.py", line 21, in <module>
    @dataclass(frozen=True)
     ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1220, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 993, in _process_class
    raise TypeError('cannot inherit frozen dataclass from a '
TypeError: cannot inherit frozen dataclass from a non-frozen one
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install through HACS version 0.2.3 on a containerized Home Assistant (?) 
2. restart HA
3. Notice the log of error

**Expected behavior**
No error and new entity created

**Additional context**
I just did the simplest path as presented on the README using HACS.  I do not know yet what is going wrong...

suaveolent, 2024-05-15:
Thanks, this might be an issue with python 3.11. 

I will have a look at it.

elafontaine, 2024-05-15:
I updated my base image and the error disappeared.  So probably python 3.11 causing some issue...

closed 2024-05-15 by elafontaine