suaveolent/ha-hoymiles-wifi #39: Total energy of panels drop when inverters goes offline

elafontaine, 2024-05-16:
**Describe the bug**
After the microinverters goes offline (no more power from the sun?), the integration report that the "total energy" for the inverter is what it was set to at the beginning of the day.  I assume that there is some kind of "hack" to add the total energy of the previous days to the currently generating data and that the inverter going offline make the calculation fail...

**To Reproduce**
Steps to reproduce the behavior:
1.  Configure the DC total energy port for each individual panel 
2. Wait for the inverter of this panel to go offline
3. observe the drop in energy produced (exactly the generation of the current day)

[image]

**Expected behavior**
No drop to a previous lower value

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I just installed the integration, so I may not be using it right.  Although I would still not expect a sensor to drop back to a value as old as the previous day...  I know this is the data sent by the DTU, so it's not that reliable either as there is probably no data for the day if the inverter is down...

suaveolent, 2024-05-29:
Thanks, I will have a look at it.

elafontaine, 2024-05-29:
In the mean time, I've used the Daily energy sensor and it's working pretty well.  So it's just the "total" one that isn't what I would have expected.

elafontaine, 2024-06-04:
In the mean time, I'm still using the individual panels value as it's more reliable.  But using the DTU sum would be nice... I'm also wondering if it's the DTU and will have a look later tonight if times allow.

sc-ampersand-p, 2024-06-12:
I just installed this addon (and the inverter, for that matter) yesterday. As expected (?) the inverter went offline over night, however, the "Total energy" entity does _not_ reset for me:

[image]

suaveolent, 2024-06-12:
> I just installed this addon (and the inverter, for that matter) yesterday. As expected (?) the inverter went offline over night, however, the "Total energy" entity does _not_ reset for me:
> 
> ![total energy keeps rising in spite of being offline](https://private-user-images.githubusercontent.com/63103291/338911737-6c138049-e8f8-468e-b00a-3047848d81e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyMDc0NDMsIm5iZiI6MTcxODIwNzE0MywicGF0aCI6Ii82MzEwMzI5MS8zMzg5MTE3MzctNmMxMzgwNDktZThmOC00NjhlLWIwMGEtMzA0Nzg0OGQ4MWUwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEyVDE1NDU0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA3OGFkYjkyZTBlMTBiNTU5YmYxMzQ2NDc0YTM2ZWM3NTE0ODkyZGFjZTE4YjA0ZWRlMzUwZjM3YmU3NDMxZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xEduQnCWapGA5d3lp0P_8vppZs_BYgwqadldkq-GDEk)

This is exactly the expected behaviour. Thanks.

elafontaine, 2024-06-13:
@sc-ampersand-p you may want to look at "total_daily_energy".

sc-ampersand-p, 2024-06-13:
Also only resets once per day, as expected, in spite of the DTU briefly going offline occasionally. 
[image]

elafontaine, 2024-06-13:
You're lucky, this is what I have ; 
[image]

sc-ampersand-p, 2024-06-13:
Is your DTU blocked from the Internet? Just curious about potential differences.

elafontaine, 2024-06-13:
> Is your DTU blocked from the Internet? Just curious about potential differences.

Nope, alway reporting to S-Miles.

sc-ampersand-p, 2024-06-13:
Same here. Might it be down to the solar panels being used? Or is that not a reasonable line of inquiry?

elafontaine, 2024-06-13:
The panels used are Thornova, but it doesn't matter, it's really the way the reporting is done into either the DTU, hoymile-wifi or this module.

elafontaine, 2024-06-13:
I suspect it's the DTU

suaveolent, 2024-06-13:
@elafontaine what really bugs me is that it starts from 0. Because this means that the DTU reports this state. 

You can cross check that by manually calling the hoymiles-wifi lib. 

Maybe there is some setting in the DTU?

suaveolent, 2024-06-14:
@stheid yes, that sounds like a good approach and should actually be rather streight forward to implement. 

I'm still hesitant to do so, since it might automatically "hide" other bugs in the DTU.

@elafontaine What happens if you directly connect to the DTU via the hoymiles app once the inverter goes offline? Do you also see a similar drop in the app?

suaveolent, 2024-06-14:
Also @elafontaine in the initial bug you were talking about the total energy dropping to 0. Do you still see this issue or is only the daily energy affected?

Simonh98, 2025-03-18:
I also have similar behavior. Does anyone have a solution for this?

[image]

suaveolent, 2025-03-18:
I will have a look at it. Which home assistant version are you on?

Simonh98, 2025-03-18:
Thank you! HomeAssistant version is the latest 2025.3.3

suaveolent, 2025-03-19:
@Simonh98 this should have been fixed a long time ago. Was it working before? E.g. before you updated to 2025.3.3?

Simonh98, 2025-03-19:
> [@Simonh98](https://github.com/Simonh98) this should have been fixed a long time ago. Was it working before? E.g. before you updated to 2025.3.3?

I have observed this behaviour since setting up this integration in mid-February with version 2025.2.4, sometimes more, sometimes less.

Maybe some more details:
- The drop in values does not occur for all ports
- The daily energy counters do not show this behaviour and remain constant

[image]

Here some debug output from the last drop:

[image]

suaveolent, 2025-03-19:
@Simonh98  Since you have 4 ports, I assume you have some kind of Hoymiles DTU and a non wifi enabled inverter right?

Does your DTU go offline at night? Can you maybe run the `get-real-data-new` command from the https://github.com/suaveolent/hoymiles-wifi library when the graph drops to 0 at night?

My assumption is that the DTU is indeed not offline (like with the wifi enabled inverters) but justs reports 0 or None values for the inverters, hence the drop in the graph.

Simonh98, 2025-03-20:
> [@Simonh98](https://github.com/Simonh98) Since you have 4 ports, I assume you have some kind of Hoymiles DTU and a non wifi enabled inverter right?

Yeah, right!

I simply wrote a script that calls `get-real-data-new` when the new total energy value is less than the old one. It is noticeable that during long dropovers overnight, less data is returned and interestingly outdated values of the total counters. However, the call at the drop at 18:19 returns the correct values.

[image]

[image]

| 18:19:33 | 18:56:41 |
|-----------|-----------|
| <small>device_serial_number: "XXXXA00XXXXX"<br>timestamp: 1742404778<br>ap: 1<br>firmware_version: 1<br>sgs_data: {<br>&nbsp;&nbsp;serial_number: XXXXXXXXXXXXXX<br>&nbsp;&nbsp;firmware_version: 1<br>&nbsp;&nbsp;voltage: 2374<br>&nbsp;&nbsp;frequency: 4997<br>&nbsp;&nbsp;active_power: 197<br>&nbsp;&nbsp;reactive_power: 219<br>&nbsp;&nbsp;current: 8<br>&nbsp;&nbsp;power_factor: 669<br>&nbsp;&nbsp;temperature: 241<br>&nbsp;&nbsp;crc_checksum: 56356<br>&nbsp;&nbsp;link_status: 1<br>&nbsp;&nbsp;modulation_index_signal: 327685<br>}<br>pv_data: {<br>&nbsp;&nbsp;serial_number: XXXXXXXXXXXXXX<br>&nbsp;&nbsp;port 1 → voltage: 277, current: 19, power: 54, energy_total: 422019, energy_daily: 1959<br>&nbsp;&nbsp;port 2 → voltage: 277, current: 19, power: 52, energy_total: 424424, energy_daily: 1958<br>&nbsp;&nbsp;port 3 → voltage: 280, current: 18, power: 50, energy_total: 425787, energy_daily: 1960<br>&nbsp;&nbsp;port 4 → voltage: 280, current: 19, power: 52, energy_total: 416858, energy_daily: 1958<br>&nbsp;&nbsp;error_code: 50331648<br>} </small>| <small>device_serial_number: "XXXXA00XXXXX"<br>timestamp: 1742407005<br>ap: 1<br>firmware_version: 1<br>sgs_data: {<br>&nbsp;&nbsp;serial_number: XXXXXXXXXXXXXX<br>&nbsp;&nbsp;modulation_index_signal: 720896<br>}<br>pv_data: {<br>&nbsp;&nbsp;serial_number: XXXXXXXXXXXXXX<br>&nbsp;&nbsp;port 1 → energy_total: 416271<br>&nbsp;&nbsp;port 2 → energy_total: 418672<br>&nbsp;&nbsp;port 3 → energy_total: 420016<br>&nbsp;&nbsp;port 4 → energy_total: 411092<br>} </small> |

elafontaine, 2025-03-25:
My initial investigation had shown to me that it was the DTU that was returning odd data as it sometime lost the connectivity to one of the inverter (I had 2 inverters connected to the same DTU).  This is a bug on the DTU side that is trying to report based on the request received, but it does not retain memory if there's a missing data (e.g. like a timeserie db that would be updated per inverter).

suaveolent, 2025-05-09:
Unfortunately I am swamped with work right now. But I'll keep the issue open until I have time to look at it.

suaveolent, 2026-10-13:
I’m closing this older issue. If the problem still occurs with the latest integration version and current DTU firmware, please open a new issue.

closed 2026-10-13 by suaveolent