suaveolent/ha-hoymiles-wifi #40: Bug fetching the DTU meter kind

elafontaine, 2024-05-29:
**Describe the bug**
The connection between HA and the DTU seems to drop every 5 minutes, while this is also a separate issue, I noticed that it also changed the DTU meter kind by adding a random character at the end.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to logbook
2. Look at the DTU entries that change values

**Expected behavior**
No change to the meter kind should change if it didn't change

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

elafontaine, 2024-05-29:
Another capture about the DTU status in HA
[image]

suaveolent, 2024-05-29:
Are you running your DTU in offline mode (e.g. blocked via Router / Firewall) ?

elafontaine, 2024-05-29:
I do not know what this means, but the DTU is reporting it's data to S-miles Cloud.  So I would assume it isn't in "offline" mode.

suaveolent, 2024-05-30:
I see. I was asking since the Hoymiles W-series has a similar issue when access to the Internet is blocked.

What is weird: I have implemented the CRC check in the library, so in my opinion these are the values the DTU transmits. Otherwise, there would be an error. However, I did not have a closer look yet.

I guess that you are already running the latest firmware of the DTU?

elafontaine, 2024-05-30:
Yes, and my guess is that there must be a single byte to much in the string...

elafontaine, 2024-06-02:
turns out I wasn't.  I was expecting the mobile application would had told me, but turns out it doesn't even suggest it even on the main s-miles pages  !  I updated to the latest (2.23) and that solved the issues.  So sorry for the noise.

[image]

closed 2024-06-02 by elafontaine

suaveolent, 2024-06-03:
Great, thanks for letting me know.