suaveolent/ha-hoymiles-wifi #43: Can´t connect to DTU

Sonnenwerk, 2024-06-12:
I want to connect to the DTU@ 192.168.178.77. Connection not possible.
DTU SW Version is 1.11
A "ping" works fine.
I can see no further help/advice in the doc, what to do or check next .,.

Sonnenwerk, 2024-06-12:
Ahh..solved the problem..
I was not aware, that i had to install the python lib as a prerequisite !

Sonnenwerk, 2024-06-12:
I would like to add, that the implementation also works with the HMT-2250-6T and the DTU-Lite S.

suaveolent, 2024-06-12:
> Ahh..solved the problem.. I was not aware, that i had to install the python lib as a prerequisite !

Thanks, ideally the python lib is installed automatically, but I have not figured out why sometimes it isn't. Have you installed it via HACS?

Sonnenwerk, 2024-06-12:
I installed "hoymiles" via HAC. But had to install the lib manually. Maybe you could reference this "can happen" in the docs ?