suaveolent/ha-hoymiles-wifi #44: Suppress state changes when DTU goes offline

sc-ampersand-p, 2024-06-14:
I find it strange that, when the DTU goes offline because the inverter switches off, some entities change their state. I understand that this is what the DTU is reporting, but from a user point of view, the entity changes are suggesting changes that aren't happening. 

Considering the fact that the Power Limiting feature is not safe to use, I was a bit alarmed the state allegedly changed multiple times per day (cloudy, so the inverter went offline multiple times): (screenshot will follow) 

What's downright false, but less concerning, is the version info for both the inverter's hardware and software: (screenshot to follow)

What I also find strange is that the DTU signal strength allegedly never changes, even though the DTU clearly was offline in the corresponding time frame: (screenshots to follow)

Making some changes here would clearly be in the "nice to have" category, maybe with the power limiter as the single exception. So not urgent or critical by any means.

Happy to discuss 😊

suaveolent, 2024-06-16:
Thanks, that indeed already on my to do list, since it bugs me myself.

suaveolent, 2024-06-16:
@sc-ampersand-p I have just released a pre-release version  (`0.2.4-dev`) which should cope with the version string. (It will just keep showing the same version, even though the inverter is offline).

As far as the signal strengh goes, I try to not immediately set the signal strength to 0 if the inverter cannot be reached for a couple of minutes, as inverters with blocked Internet access also cannot be reached sometimes.

Otherwise, the value is the one reported by the DTU. 
In the night, the value should drop to 0.

closed 2024-08-01 by github-actions[bot]

suaveolent, 2024-08-01:
Implemented in `v0.2.4`