suaveolent/ha-hoymiles-wifi #45: Can't connect to host

Seani08, 2024-06-15:
Hey, i read through the issues and already tried to reinstall the hoymiles addon and add the lib manually but this didnt solved my problem. I can ping the 800-2T successfully but cant get a connection.

Hoymiles is installed via hacs. i can also see it within the Fritzbox and S.Miles cloud.

HA is up2date
Core
2024.6.2
Supervisor
2024.06.0
Operating System
12.3
Frontend
20240610.0

any other ideas?

Seani08, 2024-06-15:
Gacs version is 1.34.0

suaveolent, 2024-06-16:
@Seani08 is there anything in the logs?

Otherwise, I just pushed a new pre-release version `0.2.4-dev` which adds some more logs if the DTU cannot be reached to help debugging.

derganzeralf1, 2024-07-15:
[image]

Got the same issue (and pretty much the same hardware setup :)) I got 0.24dev running, there are no logs showing up.

closed 2024-08-29 by github-actions[bot]