suaveolent/ha-hoymiles-wifi #47: "pip install hoymiles-wifi" with venv

Sonnenwerk, 2024-07-02:
I get an information that i should use venv for installing hoymiles wifi on my ha system.
The install guide just says "pip".
I just know, what venv is, but i can not say, if hoymiles wifi should be installed with it.

So this is not a feature request, but a request for clarification in the install guide, if i should use venv or not.
cu

stheid, 2024-07-02:
Which install guide are you referring to?

If you want to install this extension to your HA system, just follow the instructions in the readme via HACS
https://github.com/suaveolent/ha-hoymiles-wifi?tab=readme-ov-file#installation

stheid, 2024-07-02:
I think you got something mixed up: "hoymiles-wifi" is the library that this integration "ha-hoymiles-wifi" is based on. if you want to use the integration, you should only install the integration as stated in the readme of this repository. You do not need to install the library itself. That will be done automatically.

Sonnenwerk, 2024-07-02:
Yes, i got it mixed.

Sonnenwerk, 2024-07-02:
done

closed 2024-07-02 by Sonnenwerk