suaveolent/ha-hoymiles-wifi #50: Data is not sent to Hoymiles Cloud

refinedcranberry, 2024-07-29:
**Describe the bug**
When using the integration, Data is not sent to hoymiles cloud anymore. as soon as you drop the integration and restart HA, all data is transferred to the cloud.

**To Reproduce**
Steps to reproduce the behavior:
1. Add Integration with interval set to 60s
2. Notice that data in Cloud is missing after some time (for me it took ~2h). This can be identified by the Last-Update Timestamp on the overview page
3. Deactivate Integration to have all data sent to the cloud again

**Expected behavior**
Data is sent to Hoymiles cloud

suaveolent, 2024-07-29:
Thanks for reporting. 
can you please give me the model of your inverter and also which firmware you are running?

refinedcranberry, 2024-07-29:
Sure, the inverter is a HMS-800W-2T running v01.00.08. The inverters internal DTU is running v00.01.11

suaveolent, 2024-07-29:
Thanks, that is exactly what I am running here.
Can you maybe try setting it to 35 seconds just to see if it shows the same behavior?

refinedcranberry, 2024-07-29:
I now set it to 300 secs and it seems to be working for now. Tomorrow i will test with 35s and report back here

suaveolent, 2024-07-29:
> I now set it to 300 secs and it seems to be working for now. Tomorrow i will test with 35s and report back here

Thanks, so 60 seconds might be a bad value due to it being a multiple of 30. (We assume that the inverter is sending the data around every 30 seconds).

DeCysos, 2024-08-17:
I have also installed the extension today and am using it.
It's a pity that the update time is so slow. For me, 5 seconds would be perfect. 
In the Holymiles app (direct connection) it updates almost immediately.

In the HA extension, I left it at 35 seconds and can't really see anything unusual with regard to sending data from HMS-800w-2t to the cloud. Except that the data is probably only updated there every 15 minutes. (That doesn't work for me).

At the moment, I'm still leaving a Shelly Plus plugged into the AC output because it records the input current / consumption to the second.

What I think is great about the extension is that you can also read out every MPPT tracker there to check every string (or PV module).

Many thanks for the great extension ❤️

suaveolent, 2024-08-18:
> I have also installed the extension today and am using it. It's a pity that the update time is so slow. For me, 5 seconds would be perfect. In the Holymiles app (direct connection) it updates almost [...]

@DeCysos In my tests the inverter would not report the actual power if polled less than 30 seconds, but report the previous value. You can try, however, to change the paramter to 5 seconds in the `/.homeassistant/.storage/core.config_entries` file.

Just change the `"update_interval": 35,` to the preferred value.

Please let me know if that works for you, so I can remove the limit in the next version.

DeCysos, 2024-08-18:
> @DeCysos In my tests the inverter would not report the actual power if polled less than 30 seconds, but report the previous value. You can try, however, to change the paramter to 5 seconds in the [...]

Thank you for the information. I set it to 5 seconds for testing purposes and after restarting HA I can say that this works for me. At least in Home Assistant.
[image]

- In the S-Miles Installer app, the evaluation stops at 12 o'clock.
[image]
- It was displayed in the S-Miles Home app until 1:30 pm
[image]
- In the S-Miles Enduser App it was difficult to get a wifi connection, but in the end the data was displayed up-to-date.
[image]

In principle, I would almost want to leave it as it is, but I will test whether something changes with the direct connection to the DTU if I change the value in HA back to 35 seconds. If nothing changes, I will probably set the update to 10 seconds for the time being.

Here is a picture of HA's own energy dashboard of the very small PV system in cloudy weather. You can see here that the values between the HMS inverter and the Shelly socket do differ. For whatever reason, I actually calculate with the value that Shelly gives me because the value from Holymiles is probably without power loss.
[image]

DeCysos, 2024-08-22:
Update....
I had set the values to 10 seconds. 
On the following day, no more values were sent to the cloud. As a result, the statistics for 19.08.2024 remained empty even though energy was produced. The devices were also displayed as offline in the S-Miles Cloud. 
I mean, if you don't rely on the data or the statistics in the cloud or something, you can set the update time to less than 35 seconds, just as you need it.

However, this means that no more data is actually sent to Hoymiles.

PS: An update for the DTU is currently available for updating. My current version: V00.01.11 Could update to "V00.01.13_20240604", but I will not do this at the moment because I have not read a changelog.

suaveolent, 2024-08-22:
Thanks, that’s what I figured. 
I will disable the limit in the next version, but add a note that anything below around 31 seconds will lead to loss of cloud functionality. 

Yeah i know about the update. As soon as I have time I will update but log the update process again so that I might find any insight into how to trigger it locally.

DonniDoc, 2024-08-30:
Hello, 
for me the issue persists. It didn't appear during the day but can happen after startup of the inverter in the morning. 

Hardware:
HMS-800W-2T, V. 01.00.08; DTU, V 00.01.11

Integration:
CONFIG_VERSION = 2
...
DEFAULT_UPDATE_INTERVAL_SECONDS = 35
MIN_UPDATE_INTERVAL_SECONDS = 35
DEFAULT_CONFIG_UPDATE_INTERVAL_SECONDS = 60 * 5
DEFAULT_APP_INFO_UPDATE_INTERVAL_SECONDS = 60 * 60 * 2

Question: Is it correct, that the DEFAULT_APP_INFO_UPDATE_INTERVAL_SECONDS differs from the DEFAULT_CONFIG_UPDATE_INTERVAL_SECOND? 

I did some investigation. I can see on the managed switch, that the HMS is accessed 3 times per cycle. I took a closer look on the startups. The Hoymiles intgration carries two devices. The HMS and the DTU. Both devices went offline yesterday evening after the sun went down. This morning they woke up at different times: DTU at 06:16:33, HMS at 06:18:46. And from there there seems to be three queries per cycle. a. DTU, b. External to Hoymiles, c. HMS. 
Regarding a. and c. I can't exactly tell which of the queris is which dataset. It could also be a. HMS, b. External, c. DTU.
=> In the Hoymiles cloud I see no connection to the device.

After disabling and enabling the Hoymiles integration, I see only two queries and the external to Hoymiles occurs first.
=> Now the Hoymiles cloud shows connection and values.

From all I have seen I assume that there can be a situation where HA queries the inverter twice per cycle and the update to the cloud shall occur in between the two queries fom HA. Which probably cuases the issue.

suaveolent, 2024-09-03:
Yes that is exactly the behaviour we are experiencing if the inverter is polled too much. It depends on your update cycle you have in your config.

suaveolent, 2024-10-17:
@DonniDoc is still issue still relevant?

The reason why the DTU and the inverter seemingly wake up at different times is that there are different queries to the HMS. One targets the DTU data (which is polled every 5 minutes) then some version info/signal strengh (only polled every 2 minutes) and the inverter data (configured via UI).

There should be no request to the Hoymiles cloud from the integration itself. The only reason why the cloud does not work, is if the inverter data (configured via UI) is polled too often.

DonniDoc, 2024-10-22:
Thank you for asking and thank you for this integration and your great work @suaveolent! I have tried various settings for the update interval and am now at an interval of 450 seconds. That works well for me. Unfortunately, the Hoymiles inverter hardware/firmware does not allow an interval that makes sense for meaningful performance measurement or automation. So in the future I would rather choose a model without an integrated DTU and work with open DTU.

suaveolent, 2024-10-22:
Did you upgrade to the lastest firmware? I can poll my inverter every 35 seconds with cloud still working.

closed 2024-12-07 by github-actions[bot]

suaveolent, 2024-12-07:
@DonniDoc have a look at the latest update and the performance data option