suaveolent/ha-hoymiles-wifi #51: Can't find all my Inverters.

ange1us, 2024-07-31:
I have 3 inverters: HM-1500 and DTU Pro. When I add the Hoymiles integration and input the IP address, it shows only 1 DTU and 2 inverters.

I tried deleting and re-adding the integration, but it still shows only 2 inverters. I also tried restarting Home Assistant, but the issue persists. Could you please provide some guidance on how to resolve this? Here are the models and versions of my inverters:

[image]

[image]

My Home Assistant version is 2024.7.4.
Thank you.

suaveolent, 2024-07-31:
Sometimes the underlying DTU does not report back all inverters. I have not yet figured out why. 

Can you output the response you get from the `get-real-data-new` from the underlying library to see if it reports back all inverters?

https://github.com/suaveolent/hoymiles-wifi

ange1us, 2024-07-31:
I captured the "get-real-data-new" response as you suggested, and it looks like this. If I understand correctly, I can see 1 DTU, 2 inverters, and 1 meter. Is that right? It seems that only 2 inverters are being detected as well.

[image]

ange1us, 2024-07-31:
Thank you very much! It would be great if you could add support for the meter as well.

Regarding the S-Miles Installer app, I can't access it. I can only use the S-Miles End User app. However, in the S-Miles End User app, I can see all three inverters. 

As for capturing traffic with Wireshark, I'm not sure if I can do it because my switch is not a managed switch. I'll try to see if I can figure it out tomorrow. Thanks again!

suaveolent, 2024-08-01:
Got it. Can you give me the first 4 digits of your Meter Serial Number and the model you have?

Alright, let me know if you need any help with Wireshark. It would really help me identitfy the issue why sometimes not all inverters show up.

ange1us, 2024-08-01:
I'm not sure if I did it correctly. I tried using WireShark to capture traffic while accessing the S-Miles EndUser app on my phone. I also went into the device list and selected each device. It seems that it didn't detect the IP of the DTU or any of the inverters, but it did detect my phone's IP (192.168.1.88), and the results are shown here. I don't know which part I should be looking at or if it found anything.

[image]

As for my smart meter, it’s a Hoymiles DDSU666. I'm unsure which serial number is correct because in the S-Miles EndUser app, the meter's S/N is shown as 10C0xxxxx055, while in the get-data-real command, the S/N is shown as 1841xxxxxxx413.

Similarly, the inverters show different S/Ns in S-Miles EndUser app and in the get-data-real command. However, in Home Assistant, the S/Ns match with what is shown in S-Miles EndUser app.

suaveolent, 2024-08-02:
Thanks, that serial number from the `get-real-data`command is exatly what I need. I do some internal conversion to display the same serial as the Hoymiles App.

As for the Wireshark, unfortunately, it is gonna be a lot trickier. We are looking for communication on port 10081 (Filter: `tcp.port == 10081`). I'm not sure, however, if you need to connect to the DTU directly in order for the Hoymiles app do communicate directly with the DTU instead of retrieving the Data via the Cloud.

ange1us, 2024-08-02:
I tried following your suggestions, but the port 10081 didn't show up. Here's what I did:

1. Turned off Home Assistant.

2. Connected my PC via Ethernet and captured packets using Wireshark. My phone was connected to home Wi-Fi, and I accessed the S-Miles EndUser app and clicked on the various inverters. When I applied the filter tcp.port == 10081, nothing showed up.

3. Connected my PC via Ethernet and captured packets using Wireshark while my phone connected directly to the DTU via Wi-Fi. The S-Miles EndUser app couldn't display anything and showed "Service is not available," but I could access the network config. In the network config, it only offered a Wi-Fi configuration page for the DTU to connect to the router, with no other options. I checked Wireshark again with the tcp.port 10081 filter, but nothing was found.

4. Repeated steps 2 and 3, but instead of using Ethernet on the PC, I used Wi-Fi. Then again, nothing was found.

5. Connected my PC directly to the DTU via Wi-Fi, while the phone was connected either to the home Wi-Fi or directly to the DTU's Wi-Fi. The port 10081 still didn't appear.

I suspect, as you mentioned, that the S-Miles EndUser app might only retrieve data from the cloud, unlike the S-Miles Installer app, which can access data locally. Another possibility is that since I performed these tests at night, the inverters might have been shut down, maybe preventing Wireshark from capturing any data. I'll try again during the day.

Meanwhile, I'll also reach out to my solar installer to see if I can get an account for the S-Miles Installer app.

ange1us, 2024-08-06:
I've asked my solar provider, and unfortunately, they cannot provide me access to the S-Miles Installer app.

suaveolent, 2024-09-03:
That's unfortunate. As long as the `get-real-data-new` does not report all the inverters, there is nothing I can do in the meantime. I will keep the issue open to track it.

suaveolent, 2024-09-13:
@ange1us  I have just added support for meters. Can you try updating to `main` branch to see if it works?

You also have to "Reconfigure" your intergration once for the meter to get detected. Click on the three dots next to your config entry and select "Reconfigure".

pjunzip, 2024-09-14:
> @ange1us I have just added support for meters. Can you try updating to `main` branch to see if it works?
> 
> You also have to "Reconfigure" your intergration once for the meter to get detected. Click on the three dots next to your config entry and select "Reconfigure".

Thank you very much for the update. I am now able to see the smart meter; however, there are still some values showing as 'unknown,' and I'm not quite sure why. I appreciate your assistance.

[image]

ange1us, 2024-09-14:
> @ange1us I have just added support for meters. Can you try updating to `main` branch to see if it works?
> 
> You also have to "Reconfigure" your intergration once for the meter to get detected. Click on the three dots next to your config entry and select "Reconfigure".

Thank you very much for the update. I am now able to see the smart meter; however, some values are still displaying as 'unknown,' as shown in the attached image. I'm not quite sure why this is happening. I appreciate your help.

[image]

suaveolent, 2024-09-14:
Thanks, I will investigate today and push a new update. 

I noticed also some typos and wrong value conversions which I will fix today

suaveolent, 2024-09-14:
@ange1us , I just pushed a new version of `v0.2.5-dev`. Can you reinstall that version and check again?

ange1us, 2024-09-14:
I have installed the new version, and all the values are now appearing correctly. Thank you very much :)

[image]

suaveolent, 2024-09-14:
> I have installed the new version, and all the values are now appearing correctly. Thank you very much :)
> 
> ![Screenshot 2024-09-14 201355](https://private-user-images.githubusercontent.com/6756531/367517243-d868b506-21db-4ce4-9069-5e188bc06b3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzMjI5MzksIm5iZiI6MTcyNjMyMjYzOSwicGF0aCI6Ii82NzU2NTMxLzM2NzUxNzI0My1kODY4YjUwNi0yMWRiLTRjZTQtOTA2OS01ZTE4OGJjMDZiM2UucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MTRUMTQwMzU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDdjYWQ4OWZhZWQzNzhkMzAzYmQ5MTc1NjMzZTdmNDM5NDgzOWVmYTE3MWJhOGQxMTc2NDRiOWVjMTY1ODYyNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.vPytT99o5SMVoMkj9MkEvQ-svaxgKFgWQWgH1iDZeZw)

Great! 
Thanks for letting me know!

Prothesentraeger, 2024-09-27:
Hi, i have 10 Inverters HMT-2250-6T and a DTU-Pro S. 

with the hoymiles integration in Home Assistant i can see only 4 of 10 inverters.

suaveolent, 2024-10-02:
Thanks, this seems to be the same issue. 

Unfortunately I do not have access to these devices, so unless someone can capture traffic with wireshark between the DTU and the installer app there is not much I can do at this moment.

Toutatis64, 2024-11-01:
Hello, i have same issue on the inverters that are not captured (only 4 of 5 on my end). Will try to do wireshark capture

Toutatis64, 2024-11-04:
Doing a capture with wireshark of protobuf message, i get only 4 inverters of 5 in the response of the DTU (same as what I see on the python lib :/ ). Are there any parameters to customize to return more information ? Are they limited to a specific size ?

Toutatis64, 2024-11-04:
No, i guess the dtu sends directly data to the cloud in https through the network ? I don't know if i can intercept it or how i can do that ?! The only test I did was to launch the python script and check what the DTU returns

suaveolent, 2024-11-04:
Got it.
Unfortunately that does not help. The library already returns everything received from the DTU.

However, the phone app has a "toolkit" which should allow you to communicate directly with the DTU (this is the way this integration communicates)

In my opinion there are two options to capture this traffic:

1.) Connect your phone directly to DTU, and capture traffic via Wireshark (this might be difficult due to encryption. There are tutorials on how to capture encrypted traffic. Alternatively, turn off encryption for the DTU, if possible)
2.) Connect your computer to DTU install an Android emulator and capture traffic via Wireshark.
2.) Capture traffic via rooted Android phone (There are several options available)

Toutatis64, 2024-11-04:
Hello, 
i do not manage to capture traffic. 
However, i've found a way to get the last inverter I missed, so it will maybe be useful to understand what is going on .. 
In the file dtu.py of the python lib, in async_get_real_data_new() function, I configure the parameter of the request request.cp = 1 , and it returns the missing inverter ... (request.cp = 0 returns the initial 4 ones ).. 
That means that we must send 2 requests to get the whole list ? One with cp=0 , another with cp=1 ?
Hope it can help  ... :)

suaveolent, 2024-11-04:
Great find thank you. That really helps a lot! I will try to implement this!

Toutatis64, 2024-11-04:
Great thanks ! Let me know if you want me to test your branch updated once
implemented ! Thanks a lot

Toutatis64, 2024-11-04:
Another tip:  the number of inverters is correct today with the command app-information-data , this may be useful to understand the number of calls to do ?

suaveolent, 2024-11-05:
Thanks!
Can you maybe also post the response from the first and second `get-real-data-new` call so I can check if there are any differences?

Please only keep the first four digits of the serial numbers.

Toutatis64, 2024-11-05:
Get-real-data-new Response: 
device_serial_number: "1234XXXXXXX"
timestamp: 1730790604
active_power: 4
cp: 1
firmware_version: 1
sgs_data {
  serial_number: 2476
  firmware_version: 1
  voltage: 2387
  frequency: 5004
  active_power: 166
  current: 6
  power_factor: 1000
  temperature: 137
  warning_number: 1
  crc_checksum: 
  link_status: 1
  modulation_index_signal: -67
}
pv_data {
  serial_number: x2476
  port_number: 1
  voltage: 348
  current: 26
  power: 90
  energy_total: 25799
  energy_daily: 2
  error_code: 50921472
}
pv_data {
  serial_number: 2476
  port_number: 2
  voltage: 340
  current: 25
  power: 85
  energy_total: 25493
  energy_daily: 2
  error_code: 50331648
}
dtu_power: 776
dtu_daily_energy: 18

Toutatis64, 2024-11-05:
Get-real-data-new Response: 
device_serial_number: "1234XXXXXXX"
timestamp: 1730790742
active_power: 4
firmware_version: 1
meter_data {
  device_type: 1
  serial_number: 0886
  phase_total_power: -29
  phase_A_power: -29
  power_factor_total: -507
  energy_total_power: 7136
  energy_phase_A: 7136
  energy_total_consumed: 20604
  energy_phase_A_consumed: 20604
  fault_code: 1
  voltage_phase_A: 23760
  current_phase_A: 261
  power_factor_phase_A: -507
}
sgs_data {
  serial_number:5633
  firmware_version: 1
  voltage: 2378
  frequency: 5001
  active_power: 182
  current: 7
  power_factor: 1000
  temperature: 139
  warning_number: 1
  crc_checksum: 
  link_status: 1
  modulation_index_signal: -65
}
sgs_data {
  serial_number: 9213
  firmware_version: 1
  voltage: 2374
  frequency: 5001
  active_power: 88
  current: 3
  power_factor: 1000
  temperature: 136
  warning_number: 3
  crc_checksum: 
  link_status: 1
  modulation_index_signal: -65
}
sgs_data {
  serial_number: 1324
  firmware_version: 1
  voltage: 2375
  frequency: 5001
  active_power: 190
  current: 8
  power_factor: 1000
  temperature: 137
  warning_number: 2
  link_status: 1
  modulation_index_signal: -57
}
sgs_data {
  serial_number: 1928
  firmware_version: 1
  voltage: 2374
  frequency: 5001
  active_power: 182
  current: 7
  power_factor: 1000
  temperature: 135
  warning_number: 1
  crc_checksum: 
  link_status: 1
  modulation_index_signal: -71
}
pv_data {
  serial_number: 5633
  port_number: 1
  voltage: 350
  current: 28
  power: 98
  energy_total: 25634
  energy_daily: 2
  error_code: 50921472
}
pv_data {
  serial_number: 5633
  port_number: 2
  voltage: 346
  current: 27
  power: 93
  energy_total: 25588
  energy_daily: 2
  error_code: 50331648
}
pv_data {
  serial_number: 9213
  port_number: 1
  voltage: 348
  current: 26
  power: 93
  energy_total: 25481
  energy_daily: 2
  error_code: 50921472
}
pv_data {
  serial_number: 9213
  port_number: 2
  voltage: 4
  current: 1
  error_code: 50331648
}
pv_data {
  serial_number: 1324
  port_number: 1
  voltage: 351
  current: 28
  power: 101
  energy_total: 26144
  energy_daily: 2
  error_code: 50921472
}
pv_data {
  serial_number:1324
  port_number: 2
  voltage: 349
  current: 28
  power: 100
  energy_total: 25712
  energy_daily: 2
  error_code: 50331648
}
pv_data {
  serial_number: 1928
  port_number: 1
  voltage: 350
  current: 28
  power: 99
  energy_total: 25493
  energy_daily: 2
  error_code: 50921472
}
pv_data {
  serial_number: 1928
  port_number: 2
  voltage: 344
  current: 27
  power: 93
  energy_total: 25145
  energy_daily: 2
  error_code: 50331648
}
dtu_power: 816
dtu_daily_energy: 18

suaveolent, 2024-11-05:
Thanks! It will take me a while to restructure the code. But that really helps a lot!

suaveolent, 2024-11-05:
Can you maybe also post the `app-information-data` response?

suaveolent, 2024-11-05:
@Toutatis64 If I set CP to 0 for my inverter, I get the right response, but cp in the response is set to 255. What happens if you set your cp, to e.g. 255? Do you get a combined response or still only the first four inverters?

Toutatis64, 2024-11-05:
App-information-data Response: 
device_serial_number: "1234XXXXXXX"
timestamp: 1730827980
device_number: 5
pv_number: 10
package_number: 1
channel: 1
dtu_info {
  dtu_sw_version: 534
  dtu_hw_version: 38401
  dtu_step_time: 15
  dtu_rf_hw_version: 1282
  access_model: 1
  communication_time: 1730827960
  signal_strength: 100
  wifi_version: "2.3.0.0-dev"
  dtu_rule_id: 21
  sub1g_frequency_band: 2
  sub1g_channel_total_number: 21
  sub1g_channel_number: 33
  sub1g_rp: 20
  sub1g_channel_total: 10
}
meter_info {
  meter_serial_number: 0886
  meter_model: 8
  communication_way: 1
}
pv_info {
  pv_serial_number: 5633
  pv_sw_version: 10201
  pv_hw_part_number: 270627091
  pv_hw_version: 512
  pv_grid_profile_code: 3332
  pv_grid_profile: 8192
  pv_rf_hw_version: 84017154
  pv_rf_sw_version: 5
}
pv_info {
  pv_serial_number: 9213
  pv_sw_version: 10201
  pv_hw_part_number: 270627091
  pv_hw_version: 512
  pv_grid_profile_code: 3332
  pv_grid_profile: 8192
  pv_rf_hw_version: 84017154
  pv_rf_sw_version: 5
}
pv_info {
  pv_serial_number: 1324
  pv_sw_version: 10201
  pv_hw_part_number: 270627091
  pv_hw_version: 512
  pv_grid_profile_code: 3332
  pv_grid_profile: 8192
  pv_rf_hw_version: 84017154
  pv_rf_sw_version: 5
}
pv_info {
  pv_serial_number: 1928
  pv_sw_version: 10201
  pv_hw_part_number: 270627091
  pv_hw_version: 512
  pv_grid_profile_code: 3332
  pv_grid_profile: 8192
  pv_rf_hw_version: 84017154
  pv_rf_sw_version: 5
}
pv_info {
  pv_serial_number: 2476
  pv_sw_version: 10201
  pv_hw_part_number: 270627091
  pv_hw_version: 512
  pv_grid_profile_code: 3332
  pv_grid_profile: 8192
  pv_rf_hw_version: 84017154
  pv_rf_sw_version: 5
}

Toutatis64, 2024-11-05:
If I set cp to 255, empty response : 

Get-real-data-new Response: 
device_serial_number: 
timestamp: 1730828096
active_power: 4
cp: 255
firmware_version: 1
dtu_daily_energy: 9640

suaveolent, 2024-11-07:
@Toutatis64 unfortunately I noticed that package number is set to 1 for my inverter too. So I need to figure out another reliable way to query the cp parameter. Might take a bit.

Toutatis64, 2024-11-07:
Hi, 
The device_number is not relevant to increment the cp while you don't have reached the number of devices ?

suaveolent, 2024-11-07:
Yes that would be one approach, but I was hoping there was a more elegant way.

suaveolent, 2024-11-18:
@Toutatis64 I think I finally might have found the right parameter. Apparently the first `active_power` (which was named `ap`) in the mobile app is the number of packages for `real-data-new`.

In your case there should be `4` requests necessary. Can you confirm that?

Toutatis64, 2024-11-20:
Hi @suaveolent  , sorry, i don't get your point ? 
If I have device_number = 5 and active_power = 4 , that means the first request with cp=0 will return the 4 first packages, and the cp = 1 , the 4 next ones. 
So in my case, that's why, with cp = 1 , I get the last one only. 
So in my understanding, only 2 requests are necessary : math.ceil( device_number / active_power ) ?

suaveolent, 2024-11-20:
What do you get for cp = 2?

I was assuming that ap (which I falsely labeled as active_power) tells me the to total number of pages i need to request.

So in the case 4 i would request: cp 0, 1, 2, 3.

Toutatis64, 2024-11-20:
No , cp = 2 is empty for me.

Reponse is:
device_serial_number: "xxx"
timestamp: 1732108509
active_power: 4
cp: 2
firmware_version: 1
dtu_power: 1703
dtu_daily_energy: 1416

suaveolent, 2024-11-20:
That's interesting. In the end my approach will work but might issue too many requests. 

Which inverters do you have and are all ports in use?

suaveolent, 2024-12-06:
Fixed in `v0.3.3`

closed 2024-12-06 by suaveolent