suaveolent/ha-hoymiles-wifi #53: ha-hoymiles-wifi with HA in docker does not work

JoernSchimmelpfeng, 2024-08-13:
**Describe the bug**

I use HA in docker with Hacs. When adding the integration I get a HTTP 500 without more explanation. I tried to install via pip inside the container, but got an error:

> pip install ha-hoymiles-wifi 
> ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/.local'
> Check the permissions.

This obviously does not work in the container.

WORKAROUND:
Create a new HA docker image with ha-hoymiles-wifi:

> FROM homeassistant/home-assistant
> 
> RUN pip install hoymiles-wifi

Then build new docker image and switch to it:

> docker build -t ha-hoymiles

**To Reproduce**

**Expected behavior**

This workaround should be part of the documentation.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser
 - Version 

**Additional context**

suaveolent, 2024-08-18:
`pip install ha-hoymiles-wifi` this command cannot work --> you can only install the lib without the `ha` prefix. 

`pip install hoymiles-wifi`

Or did I miss something?

JoernSchimmelpfeng, 2024-09-27:
> `pip install ha-hoymiles-wifi` this command cannot work --> you can only install the lib without the `ha` prefix.
> 
> `pip install hoymiles-wifi`
> 
> Or did I miss something?

Sorry did not see your question earlier. 

You are right, that the error is from a test. I updated it with the correct name. However the result is the same.

suaveolent, 2024-09-27:
@JoernSchimmelpfeng  the permission error for the seems to be okay since you are using docker.

About the HACS error, can you install any other custom component or do you also get the `500` error?
Is there anything in the logs for this error?

JoernSchimmelpfeng, 2024-09-27:
@suaveolent The 500 is a followup problem because you cannot install the hoymiles-wifi package. As you said correctly this is because it is running in a container. This is why I suggest to provide the workaround above as part of the documentation.

suaveolent, 2024-10-17:
Thanks, got it. I will update the documentation.

closed 2024-10-17 by suaveolent