suaveolent/ha-hoymiles-wifi #54: Parameter output power (persistent or non-persistent)

RetoLu, 2024-08-19:
The output power of the inverter can be set via the openDTU with two different parameters: "permanent" (persistent) and "non-permanent" (non-persistent).
There is only one parameter, is it permanent or non-permanent. 
If the permanent parameter is selected, the power is saved permanently in the inverter's internal EEPROM memory. For technical reasons, this type of memory has a limited number of write cycles.
Would it be possible to have both parameters?
See also under: https://martinteske-blog.de/home-assistant-nulleinspeisung-balkonkraftwerk/

Tomblin, 2024-08-20:
I was wondering about the same and tried to follow it up in the code, I ended up on a constant value 8 added to a request indicating that this is a power limit change request in 
[](https://github.com/suaveolent/hoymiles-wifi/blob/f595e64303b8d15adc48a09b1a3caa2d06988fb4/hoymiles_wifi/const.py#L44)

If I understand correctly it is derived from this [](https://github.com/henkwiedig/Hoymiles-DTU-Proto/blob/main/SetPowerLimit.py) 

But I'm none the wiser which limit is set, probably the transient one, but I don't knw

suaveolent, 2024-08-20:
TL;DR: It sets it via the permanent parameter. As of this moment, there is no way we know of to only set it temporarily. 

Long answer:
See also this dicussion:
https://github.com/suaveolent/ha-hoymiles-wifi/issues/16

The parameters and requests I took in the `const.py` file are directly copied from the Android S-Miles apk.

Unfortuntatley I did not yet find a solution to only temporarily set the power limit.

suaveolent, 2024-08-20:
Maybe the parameters `CMD_ACTION_LIMIT_POWER_PF`or `CMD_ACTION_LIMIT_POWER_REACTIVE` could help here, but I personally did not test it or find anything usefule in the code.

loraerial, 2024-12-25:
I would be interested to use the Parameter Outpur power.
Do you know if there is a solution to write in non-persistent ?

Utini2000, 2025-03-11:
I am also highly interested in this since this would enable us to dynamical balance power output of any to the hoymiles connected battery without having to interact with the battery at all.

Although I am not sure if the mentioned 30 second update limit would prevent this from working anyway?

My plan was to get my apartments current power usage via Shelly 3EM every 1-2 seconds and then tell the hoymiles via HomeAssistant to increase/decrease the power output depending on the load.

suaveolent, 2025-03-11:
@Utini2000 Unfortunately for now, there only seems to be the one mode I have already implemented. I also have not found a different feature in the app.

For the Update interval, you can check out the experimental "Performance data mode" feature (you need to trigger it each day).

Utini2000, 2025-03-11:
@suaveolent aw crap. But just for my understanding: The opendtu stuff for non-W HMS models can write at the non-persistant EEPROM part?

suaveolent, 2025-03-12:
Yes , opendtu has two options. However, since the W-series communicates over a different protocol, we cannot directly apply the same strategy. If hoymiles adds such an option via the protobuf interface, we can of course also add it for this integration.

WerWolf-HH, 2025-03-17:
@suaveolent 
>> If hoymiles adds such an option via the protobuf interface, we can of course also add it for this integration.

Are you in Contact with Hoymiles about this?
Many thanks, so far, for your all effort :)

heinihuber, 2025-07-27:
I have already given up (sorry for that!) trying to get hoymiles support to implement this feature to their cloud protocol. They don't seem to be interested in dealing with our request, instead they reply with incomprehensible text modules and then simply close the case.

@suaveolent: Thank you very much for your efforts to bring this inverter to a reasonable use!

kiteman44, 2026-10-11:
Can you please take a look at this?
https://github.com/suaveolent/hoymiles-wifi/issues/108#issuecomment-6054017025

"Apparently, a new command ("action 211") has been implemented that allows setting a non-persistent power limit in watts, without writing to flash/EEPROM.

According to the developer, the limit takes effect immediately and resets to the permanently stored limit after a restart.

The implementation can be found here:
https://github.com/Eistee82/ioBroker.hoymiles/pull/85"

suaveolent, 2026-10-13:
@kiteman44 should be implemented. You can test until a proper release is made, by updating your integration to the current `main` branch.

closed 2026-10-13 by suaveolent

folfy, 2026-10-14:
@suaveolent Thanks a lot, I'm testing the new version - Overall seems to be working, just the readback for the persistent power limit seems to get messed up with values from the temporary limit. I think I noticed some special consideration for that in the ioBroker extension, seems to be an oddity of the firmware.

[image]

I'm still on Inverter FW V01.02.01 / DTU FW V01.00.07 btw - I think (DTU) FW V01.01.1 is only relevant for potential issues with the encryption, and the temporary power limit seems to have been there before already, since that's working fine.

kiteman44, 2026-10-31:
> 29\. [@kiteman44](https://github.com/kiteman44) should be implemented. You can test until a proper release is made, by updating your integration to the current `main` branch.

Thanks a lot for implementing this! I've installed the current main branch to test it, but unfortunately setting the temporary power limit doesn't work with my setup: the DTU rejects the command every time.

Setup

Inverter: HMS-1000W-2T, HW H00.04.00, FW V01.02.01
DTU: built-in Wi-Fi DTU (DTUBI), HW H00.01.00, SW V00.01.11
Integration: current main branch
What happens

The new entity "Temporary power limit" (number.dtu_limite_temporaire_de_puissance) is created correctly.
Setting any value fails with: DTU rejected the power limit (error 1)
Tested values: 10 W and 500 W, same error both times.
The persistent power limit readback shows 0 / unknown, even though the inverter is producing normally.
The inverter firmware is the same as the one @folfy reported working (V01.02.01), so I suspect the difference comes from the built-in DTUBI and its firmware (V00.01.11), compared to a standalone DTU.

I'm happy to run more tests or provide debug logs if that helps. My goal is to use the temporary limit for zero-export regulation without wearing out the flash.