suaveolent/ha-hoymiles-wifi #56: Symptoms for "Sometimes the necessary lib [...] is not correctly installed"

bcutter, 2024-08-30:
This is not a bug report. It's a question pointing at

> Note: Sometimes the necessary lib (https://github.com/suaveolent/hoymiles-wifi) is not correctly installed. In this case you need to manually install the library by running the pip install hoymiles-wifi command yourself.

How do I know if this is the case?

Background: I plan to buy an inverter supported by this integration. Before doing so, want to make 100 % sure everythink will work on HA side. For this I installed the integration. Now, even prior to restarting HA, when adding the integration, the config flow comes up with:

[image]

...which looks quite good I think.

So: green light or what are the specific symptoms indicating "the necessary lib (https://github.com/suaveolent/hoymiles-wifi) is not correctly installed" ?

suaveolent, 2024-08-30:
> This is not a bug report. It's a question pointing at
> 
> > Note: Sometimes the necessary lib (https://github.com/suaveolent/hoymiles-wifi) is not correctly installed. In this case you need to manually install the library by running the pip install hoymiles-wifi command yourself.
> 
> **How do I know if this is the case?**
> 
> Background: I plan to buy an inverter supported by this integration. Before doing so, want to make 100 % sure everythink will work on HA side. For this I installed the integration. Now, even prior to restarting HA, when adding the integration, the config flow comes up with:
> 
> ![grafik](https://private-user-images.githubusercontent.com/13799156/363138966-6eeb9901-dfab-4c1b-b02a-16ecd74b6cc0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUwMTgwOTksIm5iZiI6MTcyNTAxNzc5OSwicGF0aCI6Ii8xMzc5OTE1Ni8zNjMxMzg5NjYtNmVlYjk5MDEtZGZhYi00YzFiLWIwMmEtMTZlY2Q3NGI2Y2MwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODMwVDExMzYzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNhODJkYzVhZDQ0MWU0ZjQ0MTBhYmU4NjkwNjVhYzY0ODBkNzBiMThjNTc0ZjcxNmQ5MTJmNDAzNTI4M2MwMWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sFJ6oNbeEUpfBzG4AFm0InZYQzgbaz-k9ZLdPsAPwSk)
> 
> ...which looks quite good I think.
> 
> So: green light or **what are the specific symptoms indicating "the necessary lib (https://github.com/suaveolent/hoymiles-wifi) is not correctly installed"** ?

That looks good. Otherwise you would get an error even before the config screen is shown.

closed 2024-08-30 by suaveolent

bcutter, 2024-08-30:
OK thanks. Because the original question afterwards would have been how to fix, because the mentioned command is nothing HA OS users can use.