suaveolent/ha-hoymiles-wifi #58: Second inverter has missing entities

christof62, 2024-08-31:
A DTU-Lite-S is used to connect to 2 identical HMT-2250-6T inverters. The second identified inverter has an incomplete list of entities as there are no entities for port 3 - 6.
This behaviour is reproducable with each new configuration.

Here the entities of the second inverter:
[image]

And here the first inverter:
[image]

suaveolent, 2024-09-03:
Thanks, this seems like a similar problem as 

https://github.com/suaveolent/ha-hoymiles-wifi/issues/51

I have not yet figured out why that is. 
Do you have access to the s miles installer app and are you able to capture network traffic via wireshark?

christof62, 2024-09-04:
> Do you have access to the s miles installer app and are you able to capture network traffic via wireshark?

Yes, but unfortunately not now. This will be possible in about 2 weeks. I am not in the same network as the DTU.

suaveolent, 2024-09-04:
That’s great. Maybe you can capture the network traffic sent by the phone to see what is different. I assume you get live data of all your inverters in the app.

suaveolent, 2024-12-06:
Fixed in `v0.3.3`

closed 2024-12-06 by suaveolent