suaveolent/ha-hoymiles-wifi #59: Installation description stop at step 9 and step 10 isnt described

steinjer, 2024-09-02:
I installed ha-hoymiles-wifi with your installation steps at page https://github.com/suaveolent/ha-hoymiles-wifi. After step 9 "Click "ADD INTEGRATION" and select the Hoymiles integration." I get an pop up "Hoymiles Connection" with fields host <empty> and intervall "35 seconds" are filled. But I have no idea what to insert in host. IP adress of my dtubi.. doesn't work. 

Is this an error of my configuration or a leck in description?

robfuncken, 2024-09-02:
@steinjer i also ran into the issue, you need to find out the internal IP address of the Hoymiles device, and fill that in as the host. I looked in my router in the devices tab, it was named as Espressif.

steinjer, 2024-09-06:
> IP adress of my dtubi.. doesn't work.
In the meantime I was able to address the hoymiles by ip address. It was probably already too late for the configuration and the sun wasn't shining. Nevertheless, it would be nice if step 10 were included in the description.

suaveolent, 2024-09-06:
Thanks, I updated the Readme.md to explain the different config options.

commit 4dcb25f7 (2024-09-06): updated README.md, fixes #59
closed 2024-09-06 by suaveolent