suaveolent/ha-hoymiles-wifi #61: No way to change IP address

thomasbindl, 2024-09-06:
Since moving to a different Router the IP subnet changed and unfortunately I have to change the IP from 192.168.2.11 to 192.168.1.11, which is currently not possible, since the IP address can't be changed. 

When adding it as new entry, I get:
[image]

suaveolent, 2024-09-06:
Thanks, reconfiguring the ip address is already on the roadmap.

In the meantime you can navigate to `/.homeassistant/.storage/` and manually edit the `core.config_entries` to change the ip address.

suaveolent, 2024-09-13:
@thomasbindl I have just added support to Reconfigure the config entry.

Can you update to `main` branch to test if the feature works for you? You then need to click on the three dots next to the config entry and select "Reconfigure".

suaveolent, 2024-09-15:
Closing with release of `v0.2.5`

closed 2024-09-15 by suaveolent