suaveolent/ha-hoymiles-wifi #63: Inverter not accepting On/OFF buttons

winty87, 2024-09-13:
**Describe the bug**
Sometimes the buttons to turn the device on doesn't work, I cant turn the device on. Last night, my automation to turn the device off also didn't work which made my battery shut off via bms.

Not sure if it has something to do with the integration. Removing and readding the device doesn't help-

All sensor values like voltage, current, power setting and power are sent correctly to HA.

**Expected behavior**
Device should immediately start working after pushing the on button

**Desktop (please complete the following information):**
HASSOS

suaveolent, 2024-09-13:
Thanks @winty87.

Is there anything in the logs which might explain the issue? 
Do you manually press the buttons or do you have an automation set up?

winty87, 2024-09-14:
I am not sure if this was related to the integration but after disconnecting the inverter from both, AC and DC and reconnecting it, it seems to work again now. If this happens again in the near future I will post some log files her. Sorry that I ask but where exactly do I find the log files for the integration? Do I have to enable Debug log?

Regarding your question: Both manual and automated trigger.

suaveolent, 2024-09-15:
Thanks for letting me know. You should be able to get the log files directly from the Home Assistant log. However, you might need to enable Debug logging in your `configuration.yaml` for any usable information:

`
logger:
  logs:
    custom_components.hoymiles_wifi: debug
    hoymiles_wifi: debug
`

suaveolent, 2024-09-15:
Closing the issue for now since it seems to be resolved.

closed 2024-09-15 by suaveolent

Heppi911, 2024-09-17:
I had the same problem always acting out of an automation. I guess my problem was, I was enabling AND setting the level of power right after each other. That did not work. Now, setting a 1 s delay between the 2 commands, it seems to work.

pitrade, 2024-09-23:
I have the same problem. Adding a delay did not help