suaveolent/ha-hoymiles-wifi #67: After reboot HA no data from DTU

vmueller0206, 2024-09-23:
After rebooting HomeAssistant, no more data is read from the DTU-Pro-S. After reconfiguration, reading works again.
HA runs as a Docker container.
It doesn't matter if I restart the container or just HA.
[image]

suaveolent, 2024-09-24:
Is there anything in the logs which might explain the issue?

vmueller0206, 2024-09-24:
This ist the Docker Log after restart HA.
Nothing other Errors.

2024/09/24 12:49:44 | stdout | IndexError: list index out of range
-- | -- | --
2024/09/24 12:49:44 | stdout | ~~~~~~~~^^^
2024/09/24 12:49:44 | stdout | serial_number = getattr(sgs_data[0], "serial_number", None)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/entity.py", line 95, in __init__
2024/09/24 12:49:44 | stdout | super().__init__(config_entry, description, coordinator)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/binary_sensor.py", line 74, in __init__
2024/09/24 12:49:44 | stdout | HoymilesInverterSensorEntity(entry, updated_description, data_coordinator)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/binary_sensor.py", line 58, in async_setup_entry
2024/09/24 12:49:44 | stdout | await asyncio.shield(awaitable)
2024/09/24 12:49:44 | stdout | File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 364, in _async_setup_platform
2024/09/24 12:49:44 | stdout | Traceback (most recent call last):
2024/09/24 12:49:44 | stdout | 2024-09-24 10:49:44.782 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up hoymiles_wifi platform for binary_sensor
2024/09/24 12:49:44 | stdout | IndexError: list index out of range
2024/09/24 12:49:44 | stdout | ~~~~~~~~^^^
2024/09/24 12:49:44 | stdout | serial_number = getattr(sgs_data[0], "serial_number", None)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/entity.py", line 95, in __init__
2024/09/24 12:49:44 | stdout | super().__init__(config_entry, description, coordinator)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/sensor.py", line 566, in __init__
2024/09/24 12:49:44 | stdout | super().__init__(config_entry, description, coordinator)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/sensor.py", line 691, in __init__
2024/09/24 12:49:44 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/09/24 12:49:44 | stdout | sensor = class_name(config_entry, updated_description, coordinator)
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/sensor.py", line 550, in get_sensors_for_description
2024/09/24 12:49:44 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/09/24 12:49:44 | stdout | sensor_entities = get_sensors_for_description(
2024/09/24 12:49:44 | stdout | File "/config/custom_components/hoymiles_wifi/sensor.py", line 464, in async_setup_entry
2024/09/24 12:49:44 | stdout | await asyncio.shield(awaitable)
2024/09/24 12:49:44 | stdout | File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 364, in _async_setup_platform
2024/09/24 12:49:44 | stdout | Traceback (most recent call last):
2024/09/24 12:49:44 | stdout | 2024-09-24 10:49:44.779 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up hoymiles_wifi platform for sensor

After Reconfigure:

2024/09/24 12:57:00 | stdout | 2024-09-24 10:57:00.840 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.zahler_phasengesamtleistung (<class 'custom_components.hoymiles_wifi.sensor.HoymilesEnergySensorEntity'>) is using native unit of measurement 'W' which is not a valid unit for the device class ('energy') it is using; expected one of ['GJ', 'MWh', 'MJ', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/suaveolent/ha-hoymiles-wifi/issues
-- | -- | --
2024/09/24 12:57:00 | stdout | 2024-09-24 10:57:00.839 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.zahler_phasengesamtleistung (<class 'custom_components.hoymiles_wifi.sensor.HoymilesEnergySensorEntity'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/suaveolent/ha-hoymiles-wifi/issues

vmueller0206, 2024-09-24:
I think that's the problem. DeviceClass should be Power instead of Energy?

    HoymilesSensorEntityDescription(                                             
        key="meter_data[<meter_count>].phase_total_power",                       
        translation_key="phase_total_power",                                     
        native_unit_of_measurement=UnitOfPower.WATT,                             
        device_class=SensorDeviceClass.ENERGY,                                   
        state_class=SensorStateClass.MEASUREMENT,                                
        conversion_factor=0.1,

suaveolent, 2024-09-24:
Thanks, that helps alot.
Apparently it is trying to set up a one phase inverter, but since you don't have one it fails. I need to investigate why that is.

Can you maybe post the config entry for the integration?

You can find it in the `core.config_entries`. file in `.homeassistant/.storage/`.

vmueller0206, 2024-09-24:
"created_at": "2024-09-24T11:25:07.662921+00:00",
        "data": {
          "host": "192.168.2.201",
          "update_interval": 35,
          "dtu_serial_number": "4121********",
          "inverters": [],
          "three_phase_inverters": [
            "1382********"
          ],
          "ports": [
            {
              "inverter_serial_number": "1382********",
              "port_number": 1
            },
            {
              "inverter_serial_number": "1382********",
              "port_number": 2
            },
            {
              "inverter_serial_number": "1382********",
              "port_number": 3
            },
            {
              "inverter_serial_number": "1382********",
              "port_number": 4
            },
            {
              "inverter_serial_number": "1382********",
              "port_number": 5
            },
            {
              "inverter_serial_number": "1382********",
              "port_number": 6
            }
          ],
          "meters": [
            "10c0********"
          ]
        },
        "disabled_by": null,
        "domain": "hoymiles_wifi",
        "entry_id": "01J8HWEA6E****BDQ319QF0Q",
        "minor_version": 1,
        "modified_at": "2024-09-24T11:25:07.662924+00:00",
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "title": "192.168.2.201",
        "unique_id": "4121********",
        "version": 2

suaveolent, 2024-09-24:
Fixed with `v0.2.6-dev`

closed 2024-09-24 by suaveolent