suaveolent/ha-hoymiles-wifi #68: DTSU666 only show phase A

vmueller0206, 2024-09-23:
DTSU0666 only shows phase A.
Unfortunately, phase B, C and total energy are not displayed.
The reading is done via a DTU-Pro-S.
An HMT-2250-6T is connected as a micro inverter.
[image]

suaveolent, 2024-09-24:
Thanks, can you maybe post the output from the https://github.com/suaveolent/hoymiles-wifi `get-real-data-new` request?

Please only keep the first four digits of your serial numbers.

vmueller0206, 2024-09-24:
Get-real-data-new Response:                                                      
device_serial_number: "4121********"                                             
timestamp: 1727174120                                                            
active_power: 4                                                                  
firmware_version: 1                                                              
meter_data {                                                                     
  device_type: 3                                                                 
  serial_number: 1841**********                                                  
  phase_total_power: -28                                                         
  phase_A_power: -19                                                             
  phase_C_power: -9                                                              
  power_factor_total: -601                                                       
  energy_total_power: 35549                                                      
  energy_phase_A: 265                                                            
  energy_phase_B: 3165                                                           
  energy_phase_C: 32119                                                          
  energy_total_consumed: 71773                                                   
  energy_phase_A_consumed: 17637                                                 
  energy_phase_B_consumed: 27165                                                 
  energy_phase_C_consumed: 26971                                                 
  fault_code: 1                                                                  
  voltage_phase_A: 22330                                                         
  voltage_phase_B: 22290                                                         
  voltage_phase_C: 22340                                                         
  current_phase_A: 93
  current_phase_B: 73                                                            
  current_phase_C: 46                                                            
  power_factor_phase_A: -937                                                     
  power_factor_phase_B: 2                                                        
  power_factor_phase_C: -891                                                     
}                                                                                
tgs_data {                                                                       
  serial_number: 2145**********                                                  
  firmware_version: 1                                                            
  voltage_phase_A: 2240                                                          
  voltage_phase_B: 2230                                                          
  voltage_phase_C: 2237                                                          
  voltage_line_AB: 3864                                                          
  voltage_line_BC: 3872                                                          
  voltage_line_CA: 3881                                                          
  frequency: 4998                                                                
  active_power: 6142                                                             
  current_phase_A: 91                                                            
  current_phase_B: 91                                                            
  current_phase_C: 91                                                            
  power_factor: 1000                                                             
  temperature: 368                                                               
  warning_number: 3                                                              
  link_status: 1                                                                 
  modulation_index_signal: -7
}                                                                                
pv_data {                                                                        
  serial_number: 2145**********                                                  
  port_number: 1                                                                 
  voltage: 306                                                                   
  current: 350                                                                   
  power: 1074                                                                    
  energy_total: 55835                                                            
  energy_daily: 259                                                              
  error_code: 50986752                                                           
}                                                                                
pv_data {                                                                        
  serial_number: 2145**********                                               
  port_number: 2                                                                 
  voltage: 306                                                                   
  current: 350                                                                   
  power: 1073                                                                    
  energy_total: 53711                                                            
[... cut ...]
  power: 1033                                                                    
  energy_total: 46304                                                            
  energy_daily: 260                                                              
  error_code: 50986752                                                           
}                                                                                
dtu_power: 6142                                                                  
dtu_daily_energy: 1576

vmueller0206, 2024-09-24:
I think that's a little problem. DeviceClass should be Power instead of Energy?
Unfortunately it still only shows phase A.

HoymilesSensorEntityDescription(                                             
    key="meter_data[<meter_count>].phase_total_power",                       
    translation_key="phase_total_power",                                     
    native_unit_of_measurement=UnitOfPower.WATT,                             
    device_class=SensorDeviceClass.ENERGY,                                   
    state_class=SensorStateClass.MEASUREMENT,                                
    conversion_factor=0.1,                                 

2024/09/24 12:57:00 | stdout | 2024-09-24 10:57:00.840 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.zahler_phasengesamtleistung (<class 'custom_components.hoymiles_wifi.sensor.HoymilesEnergySensorEntity'>) is using native unit of measurement 'W' which is not a valid unit for the device class ('energy') it is using; expected one of ['GJ', 'MWh', 'MJ', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/suaveolent/ha-hoymiles-wifi/issues
-- | -- | --
2024/09/24 12:57:00 | stdout | 2024-09-24 10:57:00.839 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.zahler_phasengesamtleistung (<class 'custom_components.hoymiles_wifi.sensor.HoymilesEnergySensorEntity'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/suaveolent/ha-hoymiles-wifi/issues

suaveolent, 2024-09-24:
> phase_total_power

thanks, 

> I think that's a little problem. DeviceClass should be Power instead of Energy? Unfortunately it still only shows phase A.
>
> HoymilesSensorEntityDescription( [...]

Thanks, I already fixed that in main and an updated version of `v0.2.5`.

suaveolent, 2024-09-24:
> Get-real-data-new Response: device_serial_number: "4121********" timestamp: 1727174120 active_power: 4 firmware_version: 1 meter_data { device_type: 3 serial_number: 1841********** phase_total_power: -28 phase_A_power: -19 phase_B_power: -4 phase_C_power: -9 power_factor_total: -601 energy_total_power: 35549 energy_phase_A: 265 energy_phase_B: 3165 energy_phase_C: 32119 energy_total_consumed: 71773 energy_phase_A_consumed: 17637 energy_phase_B_consumed: 27165 energy_phase_C_consumed: 26971 fault_code: 1 voltage_phase_A: 22330 voltage_phase_B: 22290 voltage_phase_C: 22340 current_phase_A: 93 current_phase_B: 73 current_phase_C: 46 power_factor_phase_A: -937 power_factor_phase_B: 2 power_factor_phase_C: -891 } tgs_data { serial_number: 2145********** firmware_version: 1 voltage_phase_A: 2240 voltage_phase_B: 2230 voltage_phase_C: 2237 voltage_line_AB: 3864 voltage_line_BC: 3872 voltage_line_CA: 3881 frequency: 4998 active_power: 6142 current_phase_A: 91 current_phase_B: 91 current_phase_C: 91 power_factor: 1000 temperature: 368 warning_number: 3 link_status: 1 modulation_index_signal: -7 } pv_data { serial_number: 2145********** port_number: 1 voltage: 306 current: 350 power: 1074 energy_total: 55835 energy_daily: 259 error_code: 50986752 } pv_data { serial_number: 2145********** port_number: 2 voltage: 306 current: 350 power: 1073 energy_total: 53711 energy_daily: 260 error_code: 50986752 } pv_data { serial_number: 2145********** port_number: 3 voltage: 303 current: 357 power: 1082 energy_total: 54280 energy_daily: 263 error_code: 50986752 } pv_data { serial_number: 2145********** port_number: 4 voltage: 303 current: 361 power: 1094 energy_total: 53897 energy_daily: 266 error_code: 50986752 } pv_data { serial_number: 2145********** port_number: 5 voltage: 304 current: 364 power: 1106 energy_total: 55275 energy_daily: 268 error_code: 50986752 } pv_data { serial_number: 2145********** port_number: 6 voltage: 304 current: 340 power: 1033 energy_total: 46304 energy_daily: 260 error_code: 50986752 } dtu_power: 6142 dtu_daily_energy: 1576

Thanks, that helps a lot. I will add these values.

suaveolent, 2024-09-24:
@vmueller0206 I have just pushed a pre-release version (`v0.2.6-dev`), which should fix both issues. Can you please test and let me know?

vmueller0206, 2024-09-24:
After reboot HA the communication works. Screenshot for DTSU is there.
Not all parameters show.

[image]

Good Work!

Best regards,
Volker

suaveolent, 2024-09-24:
Great, thanks for letting me know. Closing this issue then.

closed 2024-09-24 by suaveolent

hmsta, 2026-05-16:
ran into the same issue.. DTSU666

I've simply changed /config/.storage/core.config_entries to type 3
```
"meters":[{"device_type":3,"meter_serial_number":"10c0169xxx"}
```

but in case it helps fixing it for others:

```
$ hoymiles-wifi --host 192.168.0.16 get-real-data-new
Get-real-data-new Response:
device_serial_number: "4121xxx"
timestamp: 1778901334
active_power: 4
firmware_version: 1
meter_data {
  device_type: 1
  serial_number: 184171985xxxx
  phase_total_power: 168
  phase_A_power: 55
  phase_B_power: 44
  phase_C_power: 68
  power_factor_total: 882
  energy_total_power: 226645
  energy_phase_A: 72878
  energy_phase_B: 43173
  energy_phase_C: 110593
  energy_total_consumed: 1273263
  energy_phase_A_consumed: 488159
  energy_phase_B_consumed: 443459
  energy_phase_C_consumed: 341644
  fault_code: 1
  voltage_phase_A: 21900
  voltage_phase_B: 22780
  voltage_phase_C: 22960
  current_phase_A: 307
  current_phase_B: 238
  current_phase_C: 306
  power_factor_phase_A: 833
  power_factor_phase_B: 818
  power_factor_phase_C: 979
}
tgs_data {
  serial_number: 213098074xxxx
  firmware_version: 1
  voltage_phase_A: 2193
  voltage_phase_B: 2282
  voltage_phase_C: 2299
  voltage_line_AB: 3862
  voltage_line_BC: 3936
  voltage_line_CA: 3936
  frequency: 5000
  active_power: 9651
  reactive_power: -71
  current_phase_A: 146
  current_phase_B: 140
  current_phase_C: 139
  power_factor: 999
  temperature: 478
  warning_number: 1
  link_status: 1
  modulation_index_signal: -72
}
tgs_data {
  serial_number: 2131001xxxx
  firmware_version: 1
  voltage_phase_A: 2193
  voltage_phase_B: 2284
  voltage_phase_C: 2302
  voltage_line_AB: 3861
  voltage_line_BC: 3940
  voltage_line_CA: 3939
  frequency: 5000
  active_power: 11517
  reactive_power: -84
  current_phase_A: 175
  current_phase_B: 168
  current_phase_C: 166
  power_factor: 999
  temperature: 475
  warning_number: 1
  link_status: 1
  modulation_index_signal: -75
}
pv_data {
  serial_number: 213098074xxxx
  port_number: 1
  voltage: 423
  current: 506
  power: 2136
  energy_total: 938535
  energy_daily: 291
  error_code: 50855680
}
pv_data {
  serial_number: 213098074xxxx
  port_number: 2
  voltage: 423
  current: 498
  power: 2100
  energy_total: 913762
  energy_daily: 326
  error_code: 50855680
}
pv_data {
  serial_number: 213098074xxxx
  port_number: 3
  voltage: 361
  current: 819
  power: 2984
  energy_total: 908166
  energy_daily: 355
  error_code: 50855680
}
pv_data {
  serial_number: 213098074xxxx
  port_number: 4
  voltage: 361
  current: 813
  power: 2959
  energy_total: 881278
  energy_daily: 343
  error_code: 50855680
}
pv_data {
  serial_number: 213100189xxxx
  port_number: 1
  voltage: 410
  current: 886
  power: 3626
  energy_total: 372731
  energy_daily: 422
  error_code: 50855680
}
pv_data {
  serial_number: 21310018xxxx
  port_number: 2
  voltage: 410
  current: 576
  power: 2356
  energy_total: 232992
  energy_daily: 294
  error_code: 50855680
}
pv_data {
  serial_number: 21310018xxxx
  port_number: 3
  voltage: 412
  current: 896
  power: 3707
  energy_total: 410674
  energy_daily: 439
  error_code: 50855680
}
pv_data {
  serial_number: 213100189xxxx
  port_number: 4
  voltage: 412
  current: 594
  power: 2459
  energy_total: 292433
  energy_daily: 317
  error_code: 50855680
}
dtu_power: 21168
dtu_daily_energy: 2787
```