suaveolent/ha-hoymiles-wifi #69: comma placement in W and Wh

vmueller0206, 2024-09-24:
Another one,
sorry.
I compared the values ​​on the DTSU display with the values ​​shown. The comma placement doesn't quite match.
I fixed it for myself in the sensor.py and added it here.

i have corected the "," in the sensor.py for meters.
HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].phase_total_power",
        translation_key="phase_total_power",
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        conversion_factor=10.0,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].phase_A_power",
        translation_key="phase_A_power",
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        conversion_factor=10.0,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].phase_B_power",
        translation_key="phase_B_power",
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        conversion_factor=10.0,
        requires_device_type=DeviceType.DEVICE_TYPE_THREE_PHASE_METER,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].phase_C_power",
        translation_key="phase_C_power",
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        conversion_factor=10.0,
        requires_device_type=DeviceType.DEVICE_TYPE_THREE_PHASE_METER,
    ),
.
.
.
.
 HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_total_power",
        translation_key="energy_total_power",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_phase_A",
        translation_key="energy_phase_A",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_phase_B",
        translation_key="energy_phase_B",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
        requires_device_type=DeviceType.DEVICE_TYPE_THREE_PHASE_METER,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_phase_C",
        translation_key="energy_phase_C",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
        requires_device_type=DeviceType.DEVICE_TYPE_THREE_PHASE_METER,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_total_consumed",
        translation_key="energy_total_consumed",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_phase_A_consumed",
        translation_key="energy_phase_A_consumed",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_phase_B_consumed",
        translation_key="energy_phase_B_consumed",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
        requires_device_type=DeviceType.DEVICE_TYPE_THREE_PHASE_METER,
    ),
    HoymilesSensorEntityDescription(
        key="meter_data[<meter_count>].energy_phase_C_consumed",
        translation_key="energy_phase_C_consumed",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
        conversion_factor=10.0,
        requires_device_type=DeviceType.DEVICE_TYPE_THREE_PHASE_METER,
    ),

Best regards,

Volker

suaveolent, 2024-09-24:
Thanks, I will push an update with the correct values tomorrow.

suaveolent, 2024-10-17:
@vmueller0206 are you sure these data are correct? You are multiplying by 10? I don't have a Hoymiles meter, so I cannot test.
But sometimes it gets confusing with the commas used by German and English Home Assistant installations (dot vs comma).

vmueller0206, 2024-10-17:
Hello,
I don't mean a comma, but a period.
e.g. 0.1, replace with 10.0,

Today i install the Update.
After that the power and consumption values were wrong
Before the Update:
[image]

After Update:
[image]

I check the sensor.py and change the values.
After this the Power Value are correct.

[image]

Best regards,
Volker

suaveolent, 2024-10-18:
Hi Volker, 

thanks for your quick reply. I understand that you changed it. However, by setting it to `10` instead of `0.01` you are multiplying the values reported by the DTU by 10. (Effectively removing any decimal place).

To my knowledge the DTU reports Wh (not kWh), so typically one would need to multiply with a factor < 0 to get the right decimal place.

Are you sure *multiplying by 10* is correct?

vmueller0206, 2024-10-18:
Hello suaveolent,
would you tell me your real name?

That fits my DTSU666.
Only phase_total_power, phase_A_power, phase_B_power and phase_C_power is factor 100(from 0.1 to 10). These are watts. 
The other Wh are all factor 10. There is no conversation_factor in your source code for energy_......_power. That's definitely 1, right?

With your code I have 3.4 watts for phase B. But it would have to be about a factor of 100. 222.1V*1.82A=404Watts.
That also fits with my values. On my DTSU today the display shows 928.92 kwh for total energy consumed and 404.20 kwh for energy fed in. This is then Factor 10 with the values ​​from your source code.

Maybe it's different on a DDSU?

Best regards,
Volker

suaveolent, 2024-10-19:
Got it. That is indeed very weird. But i will change  it.

AbuAwn, 2024-10-27:
i suaveolent,

Thank you for your hard work! I was looking for a solution to improve the slow Hoymiles cloud updates, which only occur every 15 minutes. I tested your solution over the past two days with an update interval of 35 seconds, and it’s working very well.

I also appreciate the input from vmueller0206. I noticed also this small issue and, in my case, to resolve it, I created new entities with the following configuration:    

```
meter_energy_total_consumed_corrected:
  friendly_name: "Meter Energy Total Consumed Corrected"
  unit_of_measurement: 'Wh'
  value_template: "{{ (states('sensor.meter_energy_total_consumed') | float * 10) | round(2) }}"
  device_class: energy

meter_energy_total_power_corrected:
  friendly_name: "Meter Energy Total Power Corrected"
  unit_of_measurement: 'Wh'
  value_template: "{{ (states('sensor.meter_energy_total_power') | float * 10) | round(2) }}"
  device_class: energy
```
Thanks again for the amazing contribution!

Best regards,
Adnan

suaveolent, 2024-12-06:
Fixed in `v0.3.3`

closed 2024-12-06 by suaveolent