suaveolent/ha-hoymiles-wifi #70: Migration failed while update to 0.3.0

smirgol, 2024-10-02:
**Describe the bug**
I've updated to 0.3.0 a few days ago and ever since the addon isn't working anymore. In the settings it shows "Migrationsfehler" / "Migration failed". I've checked the IP, retried reconfiguration, even downgraded, but it refuses to work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings / Devices
2. Click on 'Hoymiles'
3. In the box "Needs Attention" click on the three dots in the red error message and chose "reconfigure"
4. See error "Unknown error occured"

**Expected behaviour**
It should migrate whatever there is to migrate and display the DTU parameters on my dashboard again.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry 4 with Home Assistant 2024.9.3 Core
 - Browser: Firefox 130.0.1

**Additional context**

There are two types of error messages. This seems to be the one for the migration:
```
2024-10-01 16:36:57.997 ERROR (MainThread) [root] RealData is None. Cannot connect to DTU or invalid response received!
2024-10-01 16:36:57.997 ERROR (MainThread) [homeassistant.config_entries] Error migrating entry 192.168.0.209 for hoymiles_wifi
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 943, in async_migrate
    result = await component.async_migrate_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_wifi/__init__.py", line 112, in async_migrate_entry
    print(await async_get_config_entry_data_for_host(host))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_wifi/util.py", line 28, in async_get_config_entry_data_for_host
    raise CannotConnect
custom_components.hoymiles_wifi.error.CannotConnect
```

And this seems to get triggered if I retry too fast:

```
2024-10-01 16:39:18.065 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 462, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 537, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 366, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 413, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 516, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_wifi/config_flow.py", line 140, in async_step_reconfigure
    result = await self.hass.config_entries.async_reload(entry.entry_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1957, in async_reload
    unload_result = await self.async_unload(entry_id, _lock=False)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1906, in async_unload
    raise OperationNotAllowed(
homeassistant.config_entries.OperationNotAllowed: The config entry '192.168.0.209' (hoymiles_wifi) with entry_id 'XXXXXXX' cannot be unloaded because it is in the non recoverable state ConfigEntryState.MIGRATION_ERROR
```

Using a `Hoymiles HMS-800W-2T`
Not sure if this is of importance, but I have my single solar panel attached to input No. 2 on my Hoymiles. Input No.1 is not in use.

suaveolent, 2024-10-02:
Thanks for reporting. The error message implies that the inverter cannot be reached and thus migration fails. 

Can you issue the `get-real-data-new` from the https://github.com/suaveolent/hoymiles-wifi library to see if you get a response?

smirgol, 2024-10-02:
Thank you for your feedback!

I've interpreted the error the same way, but I'm not sure what the issue is, as the inverter is online and can be reached using the hoymiles app.

I've installed that library with pip and issued the command. It returned the data without problems.

```bash
$ pip install hoymiles-wifi

Collecting hoymiles-wifi
  Downloading hoymiles_wifi-0.2.6-py3-none-any.whl.metadata (4.4 kB)
  ...

$ hoymiles-wifi --host 192.168.0.209 get-real-data-new

<...response output...>
```

smirgol, 2024-10-02:
Hm okay, I've tried the next obvious thing and deleted the integration (the devices and entities, not the addon itself) and re-added it. Now it works again. Still not sure what went wrong when migrating from the previous version, though. :)

suaveolent, 2024-10-02:
Great. I’m closing the issue for now.

closed 2024-10-02 by suaveolent