suaveolent/ha-hoymiles-wifi #71: HMS-xxxx-4T

SchaichaD, 2024-10-07:
**Is your feature request related to a problem? Please describe.**
This microinverter has 4 ports.

**Describe the solution you'd like**
Is it possible to get the values from all 4 ports?

suaveolent, 2024-10-08:
It depends. In theory it should already work with 4 ports. 

Can you paste the output from the `get-real-data-new` command from the https://github.com/suaveolent/hoymiles-wifi lib? 

Please only keep the first 4 digits of the serial numbers.

SchaichaD, 2024-10-15:
Sorry, but where can I find the file in my homeassitant installation  „Can you paste the output from the get-real-data-new command from the https://github.com/suaveolent/hoymiles-wifi lib?“?

suaveolent, 2024-10-15:
You either need to install the tool as described in the readme or run the command from within your home assistant installation (beware if you are using a venv you need to activate it first).

SchaichaD, 2024-10-15:
Sorry, but I am a beginner. I don´t know, where I can find the needed output.
I installed the hoymiles-wifi with hacs and it works. But I have a Hoymiles HMS2000-4T and I only get values for 2 ports. I would like to get the values for all 4 ports.
[image]

suaveolent, 2024-10-16:
Alright. In principle the component supports any number of ports.

The reason I need the response from the program is that I need to check why those ports are not showing up.

I assume this is a similar issue as https://github.com/suaveolent/ha-hoymiles-wifi/issues/51 and https://github.com/suaveolent/ha-hoymiles-wifi/issues/58.

In this case the DTU does not report back all the Ports. However, I have not yet found out why.

Bernd-Reibeholz, 2024-11-13:
Hello,
I have a problem with a Hoymiles integration. I have six HM-1500 WRs on a dtu-pro, all of which are found by HA. All entities of the first two are transmitted, with the third only about half of them and then fewer and fewer of the other WRs...
Among other things, the entity AC daily energy is transmitted by the DTU, but its value is permanently "0 kWh". The current production of the WRs is also occasionally displayed as "0".
Everything is transferred correctly to the HoymilesCloud. The DTU is in the attic right next to the WRs and the connection between the DTU and the WiFi is very good with over 50% signal strength.
Does anyone have an idea?

suaveolent, 2024-11-14:
See #51 i am working on a fix for that thanks to https://github.com/Toutatis64

Bernd-Reibeholz, 2024-11-21:
Proüably that might help

 

2024-11-21 22:34:28.568 ERROR (MainThread) [homeassistant.components.number] Platform hoymiles_wifi does not generate unique IDs. ID hoymiles_01JCJBXBA9G68NGJZAKA3ZSSED_limit_power_mypower already exists - ignoring number.wechselrichter_leistungsbegrenzung

2024-11-21 22:34:28.568 ERROR (MainThread) [homeassistant.components.number] Platform hoymiles_wifi does not generate unique IDs. ID hoymiles_01JCJBXBA9G68NGJZAKA3ZSSED_limit_power_mypower already exists - ignoring number.wechselrichter_leistungsbegrenzung

2024-11-21 22:34:28.568 ERROR (MainThread) [homeassistant.components.number] Platform hoymiles_wifi does not generate unique IDs. ID hoymiles_01JCJBXBA9G68NGJZAKA3ZSSED_limit_power_mypower already exists - ignoring number.wechselrichter_leistungsbegrenzung

2024-11-21 22:34:28.569 ERROR (MainThread) [homeassistant.components.number] Platform hoymiles_wifi does not generate unique IDs. ID hoymiles_01JCJBXBA9G68NGJZAKA3ZSSED_limit_power_mypower already exists - ignoring number.wechselrichter_leistungsbegrenzung

2024-11-21 22:34:28.569 ERROR (MainThread) [homeassistant.components.number] Platform hoymiles_wifi does not generate unique IDs. ID hoymiles_01JCJBXBA9G68NGJZAKA3ZSSED_limit_power_mypower already exists - ignoring number.wechselrichter_leistungsbegrenzung

suaveolent, 2024-11-22:
@Bernd-Reibeholz 

Thanks, that is a different error which should not be related but needs to be fixed.

If you update to `main` branch, do you still don't see all entities?

Bernd-Reibeholz, 2024-11-22:
Sorry, I am a newbee in HA and english.

I just deletet the Intergration and installed it again, there is no difference form e.

What do you mean with „update to main branch“?

 

Best Regards from Berlin (Germany)

Bernd

suaveolent, 2024-12-06:
Fixed in `v0.3.3`

closed 2024-12-06 by suaveolent

Bernd-Reibeholz, 2024-12-06:
Hallo, it did not work at the first time, but whe I put up the time-intervall to 120 seconds I think, I got all entrys. Reconfigure with 30 secsnds worked aswell and now I pur it to 45 senconds and keep watching it with my 6 pieces of HM1500 on a dtu-Pro

 

Thanks a lot

 

Bernd

 

Closed #71 <https://github.com/suaveolent/ha-hoymiles-wifi/issues/71>  as completed.

suaveolent, 2024-12-06:
Great, thanks for letting me know. 

I think by changing the update interval you automatically triggered the necessary reconfigure step which would query for all inverters against