suaveolent/ha-hoymiles-wifi #72: HMT-2000 or HMT-2250

IncorporateApps, 2024-10-17:
**Describe the solution you'd like**
AC inverter total values for HMT inverters are missing. I believe they are provided by the inverters, but if they are not you can just SUM all the inverter data from DC_voltage from ports 1 to 6 (in my case 4, as I only have 4 MPPT). I receive all the DC_voltage for all 4 ports so just the SUM is missing.

Also Inverter AC grid voltage is missing for HMT inverters, but that is not so important, but would be nice to have too. 

I understand that you don't have an HMT - 3Phase device so if you need any assistance please let me know what you want to include these values.

Thank you for the consideration!

IncorporateApps, 2024-10-17:
Thank you for answering. Do you have a step by step how to get that information in  Homeassistant? Is this explained somewhere?

IncorporateApps, 2024-10-17:
We have two HMSes in there too, one 1000 and one 2000, so not sure which one you are looking for, it might be the one at the bottom.

If the information is not here you can basically just use the sum of Inverter DC Power port 1 to 4 (or in the case of HMS 2250  1 to 6?) as  sensor.invert_power for HMT inverters as the DC Power port values are correct. Maybe you do that already for HMS?Thank you for the consideration!

sgs_data {

  firmware_version: 1
  voltage: 2324
  frequency: 5005
  active_power: 109
  current: 4
  power_factor: 1000
  temperature: 98
  link_status: 1
  modulation_index_signal: -68
}
sgs_data {

  firmware_version: 1
  voltage: 2316
  frequency: 5005
  active_power: 319
  current: 13
  power_factor: 999
  temperature: 101
  link_status: 1
  modulation_index_signal: -68
}
tgs_data {

  firmware_version: 1
  voltage_phase_A: 2329
  voltage_phase_B: 2327
  voltage_phase_C: 2337
  voltage_line_AB: 4030
  voltage_line_BC: 4022
  voltage_line_CA: 4062
  frequency: 5005
  active_power: 338
  current_phase_A: 4
  current_phase_B: 4
  current_phase_C: 4
  power_factor: 1000
  temperature: 168
  link_status: 1
  modulation_index_signal: -29
}
pv_data {

  port_number: 1
  voltage: 292
  current: 16
  power: 48
  energy_total: 1786
  energy_daily: 16
  error_code: 50396928
}
pv_data {

  port_number: 2
  voltage: 300
  current: 22
  power: 67
  energy_total: 1417
  energy_daily: 23
  error_code: 50396928
}
pv_data {

  port_number: 1
  voltage: 348
  current: 24
  power: 84
  energy_total: 416
  energy_daily: 30
  error_code: 50396928
}
pv_data {

  port_number: 2
  voltage: 354
  current: 23
  power: 84
  energy_total: 575
  energy_daily: 30
  error_code: 50396928
}
pv_data {

  port_number: 3
  voltage: 351
  current: 23
  power: 83
  energy_total: 73
  energy_daily: 29
  error_code: 50396928
}
pv_data {

  port_number: 4
  voltage: 349
  current: 24
  power: 84
  energy_total: 74
  energy_daily: 30
  error_code: 50396928
}
pv_data {

  port_number: 1
  voltage: 345
  current: 26
  power: 92
  energy_total: 85
  energy_daily: 32
  error_code: 50396928
}
pv_data {

  port_number: 2
  voltage: 345
  current: 24
  power: 85
  energy_total: 80
  energy_daily: 30
  error_code: 50396928
}
pv_data {

  port_number: 3
  voltage: 345
  current: 26
  power: 92
  energy_total: 84
  energy_daily: 32
  error_code: 50396928
}
pv_data {

  port_number: 4
  voltage: 345
  current: 25
  power: 87
  energy_total: 79
  energy_daily: 30
  error_code: 50396928
}

IncorporateApps, 2024-10-17:
By the way, the value DTU signal strength, is this the signal between DTU and inverters or DTU and wifi router?

IncorporateApps, 2024-10-17:
> For the total sum, I'm not sure if the integration should provide this (if not provided by the DTU) as you could always set-up a helper in Home assistant which calculates this sum.

I only need the total sum as it is in HMS (Inverter AC power). It must be provided if they provide it for HMS, I don't see the total SUM for the HMS in the dump either?
It would be easier and cleaner. Thank you!

suaveolent, 2024-10-17:
Got it. I noticed `active_power` is only provided for single phase inverters. I will fix it and push a new version.

suaveolent, 2024-10-17:
For the grid voltage this should aready be provided. For single phase inverters this is `grid_voltage` (which is actually the phase the inverter is connected to).
For the three phase inverters this is `voltage_phase_A`, `voltage_phase_B` and `voltage_phase_C`

suaveolent, 2024-10-17:
Fixed with `v0.3.1`

closed 2024-10-17 by suaveolent