suaveolent/ha-hoymiles-wifi #73: Fail to connect

AlbertvanEngen, 2024-10-18:
**Describe the bug**
I'm using a Hoymiles DTU-Pro-S that is connected via a network cable to my router.

As soon as I try to setup the Hoymiles DTU connection, I enter the IP address (found on the router and able to ping) as Host and leave the default 35 as update interval.
Then press the send button.
Then I do get a message "Failed to connect" and the host field is cleared. 

Please advise how to solve this issue 

**To Reproduce**
Steps to reproduce the behavior:
see above

**Expected behavior**
The Hoymiles installation will be connected to my Home Assistant server

**Desktop (please complete the following information):**
 - OS:  MacOS
 - Browser safari
 - Version 15

**Smartphone (please complete the following information):**
 - Device: iPhone 15
 - OS: IOS 18.0.1
 - Browser  safari

**Additional context**
Home Assistant 2024.10.3 on Raspberry PI 3

iroc-fabi, 2024-11-03:
Bei mir gleicher Fehler. DTU PRO S wird nicht gefunden. 
DTU Version V00.02.23

suaveolent, 2024-11-03:
Is there anything in the logs? 

Can you ping your DTU from your home assistant instance?

iroc-fabi, 2024-11-03:
Verbunden. 

[image]

iroc-fabi, 2024-11-03:
[image]

suaveolent, 2024-11-03:
Can you look into the log if there is anything explaining the issue?

iroc-fabi, 2024-11-03:
No entry in the log.

iroc-fabi, 2024-11-03:
Im Protokoll ist kein Eintrag

iroc-fabi, 2024-11-03:
I have now switched off DHCP on the DTU and reset the firmware to V00.02.19. 
Doesn't help either.

suaveolent, 2024-11-03:
Can you check if the `get-real-data-new` command from the hoymiles-wifi (https://github.com/suaveolent/hoymiles-wifi) lib works?

Please only keep the first four digits of the serial numbers and post the ouput if successfull.

AlbertvanEngen, 2024-11-03:
When I execute hoymiles-WiFi --host 192.168.2.2 get-real-data-new
I do get a message: no response or unable to retrieve response for get-real-data-new

Ping however is sending a reply

suaveolent, 2024-11-03:
Well that explains why the Home Assistant integration cannot connect.

Are you sure `192.168.2.2` is the correct IP-address? Can you access the web interface of the DTU via this ip-address?

AlbertvanEngen, 2024-11-03:
This is the ip address the dtu is having on my router, connected via Itp cable.
I'm not able to connect via a web browser

suaveolent, 2024-11-03:
I See. 
When you directly connect to the DTU do you get a response for the command ?

AlbertvanEngen, 2024-11-03:
When I connect my phone to the dtu WiFi I can connect via the app, it is not working via a browser

suaveolent, 2024-11-03:
Can you connect your pc to the DTU wifi and issue the `get-real-data-new` command ? (Note that the DTU will have a different ip address)

AlbertvanEngen, 2024-11-04:
When I connect my pc directly to the WiFi router of the dtu, I do get a response from the dtu with block sgs_data and pv_data

AlbertvanEngen, 2024-11-04:
I've made a WiFi connection from the HA instance directly to the dtu, now I'm able to get data from the dtu in HA. It seems that the wired Ethernet connection is not able to read data via the local wired network

suaveolent, 2024-11-04:
That is interesting. Maybe there is something else going wrong in your setup? 
I have heard from multiple sides that the connection should work via a router too.

closed 2024-11-19 by suaveolent

lukaszlk, 2024-12-24:
Hi @wouxy, did you resolve your issue?
I’m having the exact same setup and the same Failed to Login errror when inputting DTUs IP

wouxy, 2024-12-25:
> Hi @wouxy, did you resolve your issue? I’m having the exact same setup and the same Failed to Login errror when inputting DTUs IP

Hi, yes but I’m not sure how. I updated my DTU to the latest software (S miles Cloud) and did some other stuff to my home network. Then suddenly the next day, it worked. However I still have a problem with accumulative daily power. Therefore, see an other open issue here.

lukaszlk, 2024-12-25:
@wouxy Do you remember the things you did to your network? I’ve just updated my DTU, i can ping it but i cannot Access its interface directly by IP adress. I wonder if i should try something other that switching connection to WIFI.

wouxy, 2024-12-25:
You sure your update worked? Took me a couple of times through S-Miles Cloud. 
Cannot exactly replicate anymore, but port forwarded the connection port and corresponding IP on the router. Also I switched many times between Ethernet and WiFi connection on the DTU, just for trial and error. Now I see two active network IP’s from my DTU, which I hadn’t seen before. Data however is sent over Ethernet.

lukaszlk, 2024-12-25:
Looks like it worked, I’ve updated to v04.03. I’ll try to connect it thru WiFi tomorrow. Did you set up a adress reservation for your DTU?