suaveolent/ha-hoymiles-wifi #74: What is this value

PiyaRawing, 2024-10-27:
What are the values ​​of Phase A power and Phase total power? Or are they the values ​​of the phase a load current?

[image]

Ha hoymiles v.0.3.1

suaveolent, 2024-10-27:
This is the power value of the phase the meter is connected to. There might be a conversion error though. What values is your meter reporting ? 
the reason why there is a phase total power is because some meters provide information about all three phases.

PiyaRawing, 2024-11-01:
My DDSU666 meter has the following report values ​​compared to the data shown in home assistant. I am not sure if it is correct.

[image]

[image]

suaveolent, 2024-11-03:
Thanks, that helps a lot. I will adjust the values accordingly.

suaveolent, 2024-12-06:
Fixed in `v0.3.3`

closed 2024-12-06 by suaveolent