suaveolent/ha-hoymiles-wifi #75: DTU AC daily energy drop

AbuAwn, 2024-10-30:
*Describe the bug*
In the daily energy readings from the DTU AC, there is an unexpected drop in values at some point in the afternoon. This sudden decrease doesn’t correlate with actual energy usage or any identifiable change in the input data.

*To Reproduce*
Steps to reproduce the behavior:

Go to the section in the DTU AC dashboard.
Observe the energy values throughout the day.
Notice a sudden drop in the reading during the afternoon without an apparent cause.
Expected behavior
The energy value should display a continuous progression based on actual usage without abrupt drops unless caused by significant changes in consumption.

*Screenshots*
[image]

*Additional context*
This drop in the daily energy reading seems to coincide with the time when the microinverters stop producing energy. This timing may suggest an issue in how the system records or processes energy values during periods of low or zero production, potentially causing an abrupt reset or decrease in the daily energy figure.

suaveolent, 2024-10-31:
Thanks, can you tell me a little bit more about your setup? 

When the energy drops, can you maybe also post the result of the `get-real-data-new` command from the hoymiles-wifi library https://github.com/suaveolent/hoymiles-wifi? (Please only keep the first four digits of the serial numbers?)

olivierboudet, 2024-11-08:
Hello, I think I have a similar issue, energy drops to zero at regular interval.

[image]

Thanks anyway for this integration, very easy to setup !

suaveolent, 2024-11-08:
> Hello, I think I have a similar issue, energy drops to zero at regular interval.
> 
> ![Screenshot_2024-11-08-07-13-04-44_c3a231c25ed346e59462e84656a70e50.jpg](https://private-user-images.githubusercontent.com/857205/384255110-8214ebc6-d969-4d99-9d15-40d86b9ec17d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEwNDkyMjYsIm5iZiI6MTczMTA0ODkyNiwicGF0aCI6Ii84NTcyMDUvMzg0MjU1MTEwLTgyMTRlYmM2LWQ5NjktNGQ5OS05ZDE1LTQwZDg2YjllYzE3ZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwOFQwNjU1MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YTQ3ZTJkYTM1MjJjYTRhYjljMjExYzJhYzZmODllNTkxZmE2ZmM3YTA1YTU4ZDVhODMxMzg4NjIwNTkzODk5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Um_qQBxMcdsG3LvOM5KKn6r-TqS0ZPevh_2hzJo929g)
> 
> Thanks anyway for this integration, very easy to setup !

That seems to be related with the reachability of the inverter.
Have you installed the latest firmware updates? There have been some reports that the inverter looses its WiFi connection on older firmware versions.

olivierboudet, 2024-11-09:
> That seems to be related with the reachability of the inverter.
> Have you installed the latest firmware updates? There have been some reports that the inverter looses its WiFi connection on older [...]

Thanks, firmware updates do not work with S-Miles Home so no, my inverter is not up to date.

suaveolent, 2024-11-09:
Do you have an HMS-800W-2T? 

If yes, you could try the firmware update feature of the hoymiles-WiFi lib: 

https://github.com/suaveolent/hoymiles-wifi

olivierboudet, 2024-11-14:
> That seems to be related with the reachability of the inverter.
> Have you installed the latest firmware updates? There have been some reports that the inverter looses its WiFi connection on older [...]

Thank you, it is OK with the latest firmware.

suaveolent, 2024-11-14:
Great thanks for confirming that.