suaveolent/ha-hoymiles-wifi #76: [BUG] DTU AC daily energy not counting/working

fenpaws, 2024-11-02:
**Describe the bug**

The daily energy counter doesn't work/count, but I get the information from the Inverter. 

I use a DTU-W100/DTU-Lite-S with a HM-800-2T inverter. The inverter itself provides all necessary information, and I can also use that to create a helper.

**Expected behavior**
The counter should count the daily energy creation

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Firefox
 - Version 132

**Integration**
- Version: 0.3.1

suaveolent, 2024-11-03:
Can you maybe post the output of the `get-real-data-new` command form the hoymiles-wifi https://github.com/suaveolent/hoymiles-wifi library?

Please only keep the first four digits of the serial numbers.

fenpaws, 2024-11-04:
Sorry for the delay, here is the output. 

```
Get-real-data-new Response: 
device_serial_number: "4111"
timestamp: 1730754366
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 1897
  modulation_index_signal: 1114112
}
pv_data {
  serial_number: 1897
  port_number: 1
  energy_total: 65382
}
pv_data {
  serial_number: 1897
  port_number: 2
  energy_total: 60599
}
```

suaveolent, 2024-11-05:
Couldn’t be that the inverter was already offline? I am missing a lot of data here.

fenpaws, 2024-11-05:
Well i did that at 11pm so it was already dark outside, let me do it again. I thought that you might see some other data there.

```
Get-real-data-new Response: 
device_serial_number: "4111"
timestamp: 1730810329
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 1897
  firmware_version: 1
  voltage: 2327
  frequency: 4999
  active_power: 336
  current: 14
  power_factor: 1000
  temperature: 193
  warning_number: 1
  crc_checksum: 8366
  link_status: 1
  modulation_index_signal: 2687000
}
pv_data {
  serial_number: 1897
  port_number: 1
  voltage: 389
  current: 47
  power: 184
  energy_total: 67740
  energy_daily: 85
  error_code: 50331648
}
pv_data {
  serial_number: 1897
  port_number: 2
  voltage: 387
  current: 44
  power: 169
  energy_total: 62835
  energy_daily: 82
  error_code: 50331648
}
```

suaveolent, 2024-11-05:
I see thanks.

Your DTU does not provide the AC daily energy. It only provides the daily energy for the individual ports.
I can add these in the next release.

fenpaws, 2024-11-05:
Huch thats so strange, i assume that they just calculate it in the cloud then. Well, thanks for the info. You could remove it for anyone who has the same DTU, or is it possible to add a helper that does the counting local if they are not provided.

suaveolent, 2024-11-05:
Yes I will remove it for those types of DTUs.

As far as the helper goes, I guess you can always setup this helper via Home Assistant yourself.

fenpaws, 2024-11-05:
Then that's set, i think i can close the issue now :D Thanks again!

closed 2024-11-05 by fenpaws

suaveolent, 2024-11-05:
@fenpaws If you update to the latest main branch (no release yet) the entity should be removed for your device type.