suaveolent/ha-hoymiles-wifi #77: Unable to connect HMS-800W-2T

huma-meng, 2024-11-11:
**Describe the bug**
Unable to establish a connection with the DTU of an HMS-800W-2T device. The integration fails to connect to the DTU. The issue has been tested with DTU firmware versions V00.01.11 and V00.01.13.
The Home Assistant instance is located in a different VLAN from the DTU, but routing between both VLANs is enabled and tested. As a point of comparison, a Shelly Pro 3EM device on the same VLAN as the DTU connects successfully, indicating the routing setup is functional. Additionally, the DTU is reachable via the S-Miles Cloud.

**Log**
RealData is None. Cannot connect to DTU or invalid response received!

**Expected behavior**
Integrate the HMS-800W-2T and DTU successfully into Home Assistant without connection issues.

**DTU**
 - Hardware Ver.: H00.01.00
 - Software Ver.: V00.01.13
 - Device Ver.: Gen3

**HMS-800W-2T**
 - Hardware Ver.: H00.04.00
 - Software Ver.: V01.02.01
 - Device Ver.: Gen3
 - Grid Profile Ver.: 2.0.0 (DE_VDE4105_2018)

**Home Assistant**
 - OS: Home Assistant OS VM on Proxmox
 - Version: 2024.11.2

**ha-hoymiles-wifi**
 - Version: v0.3.2

**Screenshot**
[image]

suaveolent, 2024-11-11:
Thanks for reporting. 

Although routing is confirmed by you, this error typically occurs when there is some kind of issue between communication of inverter and device.

Can you try connecting directly with your inverter (integrated hotspot) to see if connection works?

Also, can confirm you wether the `get-real-data-new` command from https://github.com/suaveolent/hoymiles-wifi works directly from your device?

Routing between VLANs should not be an issue on the integration side since I am also runnning such a setup.

huma-meng, 2024-11-11:
With a notebook directly connected to the DTU hotspot, the response to the command `hoymiles-wifi --host 10.10.100.254 get-real-data-now` looks like this:
[image]

However, when the notebook is connected to the same network but on a different VLAN, the response is: No response or unable to retrieve response for get-real-data-now.
The same issue occurs when the notebook is connected to the same network and VLAN, and even when connected to the same WiFi and VLAN.

suaveolent, 2024-11-11:
That response is exactly what I would expect. So at least communication works.

I wonder if there are some changes in newer revisions of the inverter which prevents communication via WiFi networks.

Can you setup a WiFi network without VLAN to check if communication still does not work?

(Note: if the sun has set, your DTU won't be reachable and you have to wait until the next day)

huma-meng, 2024-11-12:
I set up a WiFi network without internet access, connecting only the DTU and the notebook to it. When running `hoymiles-wifi --host <IP-ADDRESS> get-real-data-now`, I received a positive response, similar to the result when connected directly to the DTU hotspot.
It looks like I’ll need to investigate my network setup further, since the connection works outside of it. Do you have any suggestions on where to start troubleshooting? I'm using TP-Link Omada hardware with VLANs and an AdGuard Home instance.

suaveolent, 2024-11-12:
Thanks that's good news.

You could look at the permissions and ports. Communication is done via TCP streams over port 10081. 
Maybe that port is blocked for your VLAN setup?

huma-meng, 2024-11-14:
The issue is resolved! The integration is now working perfectly, and all data is displayed as expected.

Turns out the problem was caused by a WiFi setting: the "Guest Network" option was checked, which, combined with an ACL rule, completely isolated the DTU from the rest of the network. So yes, this one was definitely on my end... 

Thanks so much for the help and for sharing this project!

closed 2024-11-14 by huma-meng