suaveolent/ha-hoymiles-wifi #78: HMS-400W-1T no longer reporting any data

freaktechnik, 2024-11-18:
**Describe the bug**
Since updating to 0.3.2 my HMS-400W-1T hasn't reported any data anymore. It shows up as connected, but all other sensors and diagnostics have gone to unknown for both the DTU and the Inverter device. I still see the serial number, but the type is reported as "Unknown" in the HA UI.

**Expected behavior**
The DTU and inverter should still report their metrics.

**Screenshots**
[image]
[image]

**Additional information**
I've tried to check logs, but wasn't able to find anything related to this integration.

suaveolent, 2024-11-19:
Did you update from `0.2.X` or from `0.3.X`?

The reason I am asking is that with `0.3.X` there was a config change. If you did the update at night, it might not have retrieved all the necessary data.

You could try to reconfigure the integration. Just click on the three dots next to your entry and click on "Reconfigure".

Also: Is there anything in the logs regarding this issue?

freaktechnik, 2024-11-19:
> Did you update from `0.2.X` or from `0.3.X`?

I think so.
 
> The reason I am asking is that with `0.3.X` there was a config change. If you did the update at night, it might not have retrieved all the necessary data.

Based on logged data I did the update during the day.

> You could try to reconfigure the integration. Just click on the three dots next to your entry and click on "Reconfigure". Just make sure the sun is up when trying to do this.
> 
> Also: Is there anything in the logs regarding this issue?

This is after reconfiguring I think:
```
2024-11-19 11:27:35.483 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up hoymiles_wifi platform for sensor
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/homeassistant/helpers/entity_platform.py", line 365, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/hoymiles_wifi/sensor.py", line 632, in async_setup_entry
    sensor_entities = get_sensors_for_description(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_wifi/sensor.py", line 724, in get_sensors_for_description
    dtu_type = get_dtu_model_type(serial_bytes)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/hoymiles_wifi/hoymiles.py", line 295, in get_dtu_model_type
    raise ValueError(f"Unknown DTU: {serial_bytes[:2]}!")
ValueError: Unknown DTU: b'AD'!
2024-11-19 11:27:35.484 ERROR (MainThread) [hoymiles_wifi] Unknown inverter type: 0x14 0x3
2024-11-19 11:27:35.485 ERROR (MainThread) [hoymiles_wifi] Unknown DTU: b'AD'!
2024-11-19 11:27:35.485 ERROR (MainThread) [hoymiles_wifi] Unknown DTU: b'AD'!
2024-11-19 11:27:35.486 ERROR (MainThread) [hoymiles_wifi] Unknown inverter type: 0x14 0x3
2024-11-19 11:27:35.486 ERROR (MainThread) [hoymiles_wifi] Unknown inverter type: 0x14 0x3
```

suaveolent, 2024-11-19:
That's weird. Was your inverter/DTU successfully detected before that? (Meaning under Device Info: could you see the HMS-400W-1T?

Can you maybe post the reponse of `get-real-data-new` from the https://github.com/suaveolent/hoymiles-wifi (please only keep the first 6 digits of the serial number)

freaktechnik, 2024-11-19:
> That's weird. Was your inverter/DTU successfully detected before that? (Meaning under Device Info: could you see the HMS-400W-1T?

I don't recall, but it did report data (as shown in screenshots)

```
[martin@martin-desktop ~]$ hoymiles-wifi --host 192.168.1.126 get-real-data-new
Get-real-data-new Response: 
device_serial_number: "4144A0XXXXXXXX"
timestamp: 1732012795
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 220058XXXXXXXXXX
  firmware_version: 1
  voltage: 2326
  frequency: 4998
  active_power: 101
  current: 4
  power_factor: 1000
  temperature: 114
  warning_number: 2
  crc_checksum: 28864
  link_status: 1
  modulation_index_signal: 10748101
}
pv_data {
  serial_number: 220058XXXXXXXX
  port_number: 1
  voltage: 296
  current: 36
  power: 106
  energy_total: 9144
  energy_daily: 14
  error_code: 50921472
}
dtu_power: 101
dtu_daily_energy: 14
```

suaveolent, 2024-11-19:
Thanks, might be that I introduced a regression for type `Unknown` with `0.3.X`. I will have a look and add support for HMS-400W-1T too.

suaveolent, 2024-11-19:
@freaktechnik I have pushed a quick fix for your issue, adding support for `Unknown` DTU types again. You can test it by upgrading to the `main` branch.

A full fix will be provided in the next release at it requires an update of the `hoymiles-wifi` lib.

freaktechnik, 2024-11-19:
It seems hacs only offers me actual releases/tags in the repo? Is there some way to get main via HACS?

suaveolent, 2024-11-19:
Right this has been changed in Hacs 2.0.

you need to go to "Developer Tools" - "Actions":

 - Actions: "Update: install update"
 - Targets: "Hoymiles update"
 - Version: "main" 

Afterwards there should be an Update displayed under "Settings" (Note: The version will still be `0.3.2`).

freaktechnik, 2024-11-19:
Can confirm that everything is reported again with main / f75d97c

suaveolent, 2024-11-19:
> Can confirm that everything* is reported again with main / [f75d97c](https://github.com/suaveolent/ha-hoymiles-wifi/commit/f75d97cd0c1f15d56427fff5eb5368145f035ba5)
> 
> Thank you!
> 
> (* except DTU daily energy)

That’s weird. According to your data posted it should work. I’ll have a closer look at it.

suaveolent, 2024-11-19:
A right got it: since your DTU is not correctly identified it does not show that value. Once I have updated the hoymiles-WiFi lib the DTU will be recognized as DTUBI and daily energy will work again.

freaktechnik, 2024-11-19:
> A right got it: since your DTU is not correctly identified it does not show that value. Once I have updated the hoymiles-WiFi lib the DTU will be recognized as DTUBI and daily energy will work again.

I assumed as much from the changes in the latest updates, thus only the footnote. I don't depend on that value for anything, so I'm fine without it.

suaveolent, 2024-11-19:
@freaktechnik If you again update to `main` branch you should be able to correctly see your inverter model as well as the `daily energy`.

freaktechnik, 2024-11-20:
Confirmed. Thanks again.

closed 2024-12-06 by suaveolent

freaktechnik, 2024-12-06:
Inverter type is now reported as the correct model. Thanks!

suaveolent, 2024-12-06:
Thanks for confirming 👌