suaveolent/ha-hoymiles-wifi #82: Export Power Limit

PiyaRawing, 2024-12-05:
When I enable the Export Power Limit (in the Hoymiles App) 
[image]
and enable the Hoymiles integration (in Home Assistant), I notice that power still occasionally flows back to the grid. Additionally, there are periods when the inverter stops working, dropping to 0 kW.

However, when I disable the Hoymiles integration (in Home Assistant) , the Export Power Limit functions normally without any reverse flow occurring.

I have never used the Controls Power Limit feature in the Hoymiles integration.

[image]

- **DTU Pro (S) V00.02.23**
- **HMS-2000D-4T**
- **Ha hoymiles v.0.3.2**

suaveolent, 2024-12-12:
Hm that’s interesting. What is your update interval?

PiyaRawing, 2024-12-12:
I use the default setting: 35 seconds.

Now I've updated to version 3.3.0, but it's still the same.

suaveolent, 2024-12-12:
Does this also happen if you e.g. access the DTU with the mobile app and live monitoring? 

I think you need to use the s Miles installer app for that.

PiyaRawing, 2024-12-12:
I tried both on the web and the S-miles Installer app. The result is the same.

[image]

PiyaRawing, 2024-12-13:
Realtime Monitor S-miles Installer 

[image]

[image]

suaveolent, 2024-12-13:
Thanks, I think I explained it wrong.

The S Miles Installer has a "toolkit" which allows you to observe live data (this is the exact same way the integration communicates with the DTU).

I would be curious if the same issue arises when you disable the home assistant integration and keep using the "toolkit" for a couple of hours.

PiyaRawing, 2024-12-14:
From trying to set Export Power Limit: 0 and continuously viewing data in "Toolkit" for about 2 hours, it was found that data was not sent to Cloud and was not saved in Historical data since monitoring through "Toolkit".

I also tried using a Tuya smart meter to check if the export limit was working and found that it wasn't functioning at all. The micro inverter continued operating at 100%.

[image]

[image]

suaveolent, 2025-01-23:
> From trying to set Export Power Limit: 0 and continuously viewing data in "Toolkit" for about 2 hours, it was found that data was not sent to Cloud and was not saved in Historical data since [...]

Thanks, this more or less confirms that this is an issue with the way Hoymiles implements the Export Power Limit. In the end, all that should help here is to increase your update interval (let's say every X minutes) until the issue no more occurs or report the bug to hoymiles.

closed 2025-03-10 by github-actions[bot]