suaveolent/ha-hoymiles-wifi #83: Integration disappear after reboot

Stromber03, 2024-12-05:
**Describe the bug**
After each reboot of Homeassistant,  the integration disappear 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to integration 
2. Clic add and chose Hoymiles
3. config IP
4. clic ok and add DTU and onduleur device to homeassistant 
5. reboot homeassistant
6. integration disappear 

**Expected behavioclea
Integration have to be here after reboot 

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

suaveolent, 2024-12-06:
That is weird and should not be related to the integration. 

Can you tell me more about your setup? What version of home assistant are you using? What installation method and so on…

Stromber03, 2024-12-06:
I used the Hacs module. Option 1,

I added the integration 

I added the ip, and I found DTU and Onduleur
[image]

Then I restarted Homeassistant and the entire integration disappear 

## System Information

version | core-2024.12.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.63-haos
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1540
Downloaded Repositories | 76

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 23 novembre 2025 à 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | eu-central-1-23.ui.nabu.casa
certificate_status | ready
instance_id | 16798969dd2d42599713d55afcdd197a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 219.4 GB
disk_used | 43.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.2), Home Assistant Google Drive Backup (0.112.1), Advanced SSH & Web Terminal (19.0.0), Mosquitto broker (6.4.1), Studio Code Server (5.17.3), TasmoAdmin (0.31.0), ESPHome Device Compiler (2024.11.2), Grafana (10.2.2), InfluxDB (5.0.1), Z-Wave JS UI (3.17.0), Duck DNS (1.18.0), AppDaemon (0.16.7), Zigbee2MQTT (1.42.0-2), Matter Server (6.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 50
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 novembre 2024 à 06:45
-- | --
current_recorder_run | 5 décembre 2024 à 23:00
estimated_db_size | 5762.84 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 0 / 3
local_online | 3 / 3

</details>## System Information

version | core-2024.12.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.63-haos
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1540
Downloaded Repositories | 76

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 23 novembre 2025 à 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | eu-central-1-23.ui.nabu.casa
certificate_status | ready
instance_id | 16798969dd2d42599713d55afcdd197a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 219.4 GB
disk_used | 43.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.2), Home Assistant Google Drive Backup (0.112.1), Advanced SSH & Web Terminal (19.0.0), Mosquitto broker (6.4.1), Studio Code Server (5.17.3), TasmoAdmin (0.31.0), ESPHome Device Compiler (2024.11.2), Grafana (10.2.2), InfluxDB (5.0.1), Z-Wave JS UI (3.17.0), Duck DNS (1.18.0), AppDaemon (0.16.7), Zigbee2MQTT (1.42.0-2), Matter Server (6.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 50
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 novembre 2024 à 06:45
-- | --
current_recorder_run | 5 décembre 2024 à 23:00
estimated_db_size | 5762.84 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 0 / 3
local_online | 3 / 3

</details>

suaveolent, 2024-12-06:
Let me update to 2024.12 and see if I can reproduce the issue. 
what about other integrations from hacs? Are the disappearing too?

suaveolent, 2024-12-07:
Thanks for letting me know.

closed 2024-12-07 by suaveolent

Stromber03, 2024-12-07:
I restored my server and now everything is fine