suaveolent/ha-hoymiles-wifi #84: Last daily hour accumulates daily production

wouxy, 2024-12-20:
The plugin is workin well, exept for one issue I consequently encounter every day: The 'DTU AC daily energy' value accumulates the whole production of that day in the last hour as you can see in the screenshots below. Looks like the plugin is calculating the daily production twice in the last connected hour.

DTU Pro-S
Hardware H09.06.11
Software V00.04.03
[image]
[image]

suaveolent, 2024-12-20:
I have seen a similar graph already. The issue seems to be that once the inverter goes offline the DTU reports back a different value. Once the "drop" happens, Home Assistant thinks that the sensor has been reset since and accumulates the result. This is why you see the increase. 

The question is: why does the drop happen in the first place.

Can you maybe post the reponse from the `get-real-data-new` request from the `hoymiles-wifi` lib https://github.com/suaveolent/hoymiles-wifi before and after the drop?

wouxy, 2024-12-21:
**Please not that I dont use al DC ports on one inverter: 

This is between sunset and midnight:**

TU Response: device_serial_number: "10FD********"
timestamp: 1734731511
ap: 1
firmware_version: 1
tgs_data {
  serial_number: 2145**********
  warning_number: 9
  modulation_index_signal: 24772608
}
tgs_data {
  serial_number: 2145**********
  warning_number: 3
  modulation_index_signal: -63
}
pv_data {
  serial_number: 2145**********
  port_number: 1
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 2
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 3
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 4
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 5
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 6
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 1
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 2
  error_code: 589824
}
pv_data {
  serial_number: 21451**********
  port_number: 3
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 4
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 5
  error_code: 589824
}
pv_data {
  serial_number: 2145**********
  port_number: 6
  error_code: 589824
  
  
  **And this is when the inverters are active:**
  
  DTU Response: device_serial_number: "10FD**********
timestamp: 1734791198
ap: 1
firmware_version: 1
tgs_data {
  serial_number: 2145**********
  firmware_version: 1
  voltage_phase_A: 2273
  voltage_phase_B: 2326
  voltage_phase_C: 2280
  voltage_line_AB: 3969
  voltage_line_BC: 3988
  voltage_line_CA: 3957
  frequency: 4997
  active_power: 107
  current_phase_A: 1
  current_phase_B: 1
  current_phase_C: 1
  power_factor: 999
  temperature: 102
  warning_number: 5
  link_status: 1
  modulation_index_signal: -56
}
tgs_data {
  serial_number: 2145**********
  firmware_version: 1
  voltage_phase_A: 2273
  voltage_phase_B: 2279
  voltage_phase_C: 2324
  voltage_line_AB: 3958
  voltage_line_BC: 3985
  voltage_line_CA: 3969
  frequency: 4997
  active_power: 221
  current_phase_A: 3
  current_phase_B: 3
  current_phase_C: 3
  power_factor: 1000
  temperature: 111
  warning_number: 2
  link_status: 1
  modulation_index_signal: -55
}
pv_data {
  serial_number: 2145**********
  port_number: 1
  voltage: 307
  power: 1
  energy_total: 1149
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 2
  voltage: 307
  current: 11
  power: 35
  energy_total: 583594
  energy_daily: 64
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 3
  voltage: 315
  power: 1
  energy_total: 925
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 4
  voltage: 315
  current: 11
  power: 36
  energy_total: 633209
  energy_daily: 65
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 5
  voltage: 310
  energy_total: 412
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 6
  voltage: 310
  current: 12
  power: 37
  energy_total: 663337
  energy_daily: 67
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 1
  voltage: 311
  current: 13
  power: 40
  energy_total: 662697
  energy_daily: 71
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 2
  voltage: 311
  current: 12
  power: 39
  energy_total: 661166
  energy_daily: 70
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 3
  voltage: 308
  current: 12
  power: 40
  energy_total: 651761
  energy_daily: 70
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 4
  voltage: 308
  current: 12
  power: 38
  energy_total: 645157
  energy_daily: 68
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 5
  voltage: 314
  current: 12
  power: 37
  energy_total: 608681
  energy_daily: 68
  error_code: 50986752
}
pv_data {
  serial_number: 2145**********
  port_number: 6
  voltage: 314
  current: 12
  power: 37
  energy_total: 621869
  energy_daily: 68
  error_code: 50986752
}
dtu_power: 328
dtu_daily_energy: 611

jzun-a, 2025-01-15:
Having exactly same issue. Would appreciate little effort on this. 
Unfortunately, my own knowledge is not enough to solve this...

suaveolent, 2025-01-20:
I suspect that this is related to https://github.com/suaveolent/ha-hoymiles-wifi/pull/85.

After confirming that this is really a bug with the hoymiles dtu I will merge the PR and push a new version.

suaveolent, 2025-01-27:
Should be fixed in latest release

closed 2025-01-27 by suaveolent